jurkovic-nikola/OpenLinkHub #88: OpenLinkHub service not working?

Nytem, 2025-03-15:
So the service seems to be failing? I tried updating to the latest version and it still fails. I'm not quite sure where the issue is starting from?

`simon@fedoraWorkstation:~$ sudo systemctl status OpenLinkHub
× OpenLinkHub.service - Open source interface for iCUE LINK System Hub, Corsair AIOs and Hubs
     Loaded: loaded (/usr/lib/systemd/system/OpenLinkHub.service; enabled; preset: disabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf, 50-keep-warm.conf
     Active: failed (Result: exit-code) since Fri 2025-03-14 23:32:36 EDT; 7s ago
   Duration: 2.534s
 Invocation: 943dd72cd5be4de18ff8bbbd010ff890
    Process: 224403 ExecStart=/opt/OpenLinkHub/OpenLinkHub (code=exited, status=2)
   Main PID: 224403 (code=exited, status=2)
   Mem peak: 44.6M
        CPU: 735ms

Mar 14 23:32:36 fedoraWorkstation OpenLinkHub[224403]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x5e17d9]
Mar 14 23:32:36 fedoraWorkstation OpenLinkHub[224403]: goroutine 74 [running]:
Mar 14 23:32:36 fedoraWorkstation OpenLinkHub[224403]: OpenLinkHub/src/systeminfo.GetGPUUtilization()
Mar 14 23:32:36 fedoraWorkstation OpenLinkHub[224403]:         /home/nikola/Git/OpenLinkHub/src/systeminfo/systeminfo.go:454 +0x19
Mar 14 23:32:36 fedoraWorkstation OpenLinkHub[224403]: OpenLinkHub/src/devices/cc.(*Device).setupLCD.func1()
Mar 14 23:32:36 fedoraWorkstation OpenLinkHub[224403]:         /home/nikola/Git/OpenLinkHub/src/devices/cc/cc.go:2662 +0x295
Mar 14 23:32:36 fedoraWorkstation OpenLinkHub[224403]: created by OpenLinkHub/src/devices/cc.(*Device).setupLCD in goroutine 82
Mar 14 23:32:36 fedoraWorkstation OpenLinkHub[224403]:         /home/nikola/Git/OpenLinkHub/src/devices/cc/cc.go:2552 +0xe5
Mar 14 23:32:36 fedoraWorkstation systemd[1]: OpenLinkHub.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Mar 14 23:32:36 fedoraWorkstation systemd[1]: OpenLinkHub.service: Failed with result 'exit-code'.
`

- OS: Fedora 41

I'm not sure if this has anything to do with it, but I just noticed this after swapping my GPU in my PC from my Asus ROG STRIX 3070 ti to a XFX Mercury 9070 XT.

If there's any files that may show what's causing the issue, just tell me where to look and I'll send them right here.

jurkovic-nikola, 2025-03-15:
It doesn't read GPU data properly. For the time being, you can edit profile JSON file in /opt/OpenLinkHub/database/profiles/ and edit lcdMode to 1 until it's fixed.

Nytem, 2025-03-15:
Ah that did the trick, thanks!

If there is any data I can get you regarding this GPU that would help fix the issue in a future version then I'd be happy to run some commands and send it over.

jurkovic-nikola, 2025-03-15:
No, I'll fix it once i'm back. I have AMD card for testing.

Nytem, 2025-03-15:
alrighty, thanks again for your hard work!

closed 2025-03-15 by Nytem