jurkovic-nikola/OpenLinkHub #90: memory set true not work

marcleonio, 2025-03-17:
**Describe the bug**
A clear and concise description of what the bug is.
Memory not show

**To Reproduce**
Steps to reproduce the behavior:
1. Your configuration values
2. Number of devices
3. Content of `/opt/OpenLinkHub/stdout.log`

{
    "debug": false,
    "listenPort": 27003,
    "listenAddress": "127.0.0.1",
    "cpuSensorChip": "k10temp",
    "manual": false,
    "frontend": true,
    "metrics": false,
    "memory": true,
    "memorySmBus": "i2c-7",
    "memoryType": 5,
    "exclude": [],
    "decodeMemorySku": false,
    "memorySku": "CMT64GX5M2B5200C40",
    "resumeDelay": 15000,
    "logFile": "",
    "logLevel": "info"
}

`
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is missing, creating initial one.","time":"2025-03-17T02:46:30-03:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is created.","time":"2025-03-17T02:46:30-03:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is missing, creating initial one.","time":"2025-03-17T02:46:30-03:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is created.","time":"2025-03-17T02:46:30-03:00"}
{"level":"warning","msg":"No profile found for device. Probably initial start","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:31-03:00"}
{"firmware":"1.2.265","level":"info","msg":"This firmware can support only 14 devices.","product":"iCUE LINK System Hub","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:31-03:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"level":"info","msg":"Device successfully initialized","product":"iCUE LINK System Hub","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:46:32-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:50:41-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:50:41-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:50:51-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:50:51-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:50:55-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:50:55-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:51:05-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:51:05-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:51:13-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:51:13-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:51:23-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:51:23-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:52:02-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:52:02-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:52:12-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:52:12-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:52:16-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:52:16-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:52:21-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:52:21-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:52:27-03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/777101B9D3C68E59A59CF016E9E61F94.json","msg":"Loaded custom user profile","serial":"777101B9D3C68E59A59CF016E9E61F94","time":"2025-03-17T02:52:27-03:00"}
{"devices":3,"level":"info","msg":"Device amount matches position amount.","positions":3,"time":"2025-03-17T02:52:38-03:00"}
[... 195 lines cut ...]

show memory

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

**Desktop (please complete the following information):**
 - OS: [Ubuntu 24.04 LIS]
 - Device Firmware
 - Program Version [0.5.1]
 -  [msi b650m gaming plus wifi](https://www.msi.com/Motherboard/B650M-GAMING-PLUS-WIFI)
 - DOMINATOR PLATINUM RGB	DDR5 2x32 slot DIMMA2 DIMMB2 

[image]

[image]

[image]

**Additional context**
Add any other context about the problem here.

jurkovic-nikola, 2025-03-17:
Your're doing i2c detect on i2c-2 in screenshot, but you're using i2c-7 in config. Display info from i2c-7 please.

marcleonio, 2025-03-20:
[image]

marcleonio, 2025-03-20:
[image]

[image]

it worked! thanks for the tip about searching for 7

marcleonio, 2025-03-20:
spd5118 is a kernel driver for a temperature sensor from the [Vitals plugin](https://github.com/corecoding/Vitals)
Vitals is a GNOME Shell extension for displaying your computer's temperature
I'll blacklist him

It appears disabled here, but I don't care.
[image]

jurkovic-nikola, 2025-03-20:
Unloading the module won't cause any issue. This is done by doing module blacklist like you did. This wasn't required before, but after some kernel version, registers started to be locked with UU status.

closed 2025-03-20 by marcleonio