jurkovic-nikola/OpenLinkHub #91: No devices error

szirbekur, 2025-03-18:
**Describe the bug**
A clear and concise description of what the bug is.
No device shown in dashboard

**To Reproduce**
Steps to reproduce the behavior:
1. Your configuration values
2. Number of devices
3. Content of `/opt/OpenLinkHub/stdout.log`

{
  "debug": false,
  "listenPort": 27003,
  "listenAddress": "127.0.0.1",
  "cpuSensorChip": "zenpower",
  "manual": false,
  "frontend": true,
  "metrics": true,
  "resumeDelay": 15000,
  "memory": true,
  "memorySmBus": "i2c-16",
  "memoryType": 5,
  "exclude": [],
  "decodeMemorySku": false,
  "memorySku": "CMH32GX5M2B6000Z30",
  "logFile": "",
  "logLevel": "info",
  "enhancementKits": [80,82]
}

{"file":"/home/sirbekur/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2025-03-18T22:45:30+01:00"}
{"file":"/home/sirbekur/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2025-03-18T22:45:30+01:00"}
{"file":"/home/sirbekur/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is found, checking for any upgrade...","time":"2025-03-18T22:45:30+01:00"}
{"file":"/home/sirbekur/OpenLinkHub/database/scheduler.json","level":"info","msg":"Nothing to upgrade.","time":"2025-03-18T22:45:30+01:00"}
{"error":"no devices","level":"warning","msg":"No valid I2C devices found","time":"2025-03-18T22:45:31+01:00"}

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information):**
 - OS: RaveOS GNOME 1.0  Linux 6.13.7-zen1-1-zen
 - MSI X870 Tomahawk Wifi
 - Ryzen 7 7700X
 - Corsair 32GB KIT DDR5 6000MHz CL30 Vengeance RGB Grey EXPO

**Additional context**
Add any other context about the problem here.

[image]

jurkovic-nikola, 2025-03-19:
Looks like program can't enumerate i2c devices. Will take a look on test system what's going on.

szirbekur, 2025-03-19:
> Looks like program can't enumerate i2c devices. Will take a look on test system what's going on.

Thank you

jurkovic-nikola, 2025-03-22:
Install latest 0.5.2 build and attach stdout.log file here. Let's see what's going on.

closed 2025-04-09 by jurkovic-nikola