jurkovic-nikola/OpenLinkHub #93: [Feature Request] Please add support for H115i PRO RGB 280mm AIO

tchubaba, 2025-03-21:
**Describe the solution you'd like**
I'd like to request support for the H115i PRO RGB 280mm AIO

**Device Information from `lsub`**
Bus 001 Device 005: ID 1b1c:0c13 Corsair H115i Platinum

**Additional context**
Link to the device in question from corsair.com:
https://www.corsair.com/us/en/p/refurb-cpu-coolers/cw-9060032-ww-rf/hydro-series-h115i-pro-rgb-280mm-liquid-cpu-cooler-refurbished-cw-9060032-ww-rf

jurkovic-nikola, 2025-03-21:
Unfortunately this device is so old, I can't even source it. Without physical device, implementation is not possible.

tchubaba, 2025-03-30:
Sure thing, the requested info is below:

```
lsusb -d 1b1c:0c13 -v

Bus 001 Device 005: ID 1b1c:0c13 Corsair H115i Platinum
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c13 
  bcdDevice            1.00
  iManufacturer           1 Corsair
  iProduct                2 H115i Platinum
  iSerial                 3 7289_2.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower               64mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

It seems to identify itself as "Corsair H115i Platinum" for some reason but it's definitely the H115i PRO RGB. If you need need anything else let me know!

Thanks.

tchubaba, 2025-03-30:
Well, here it is again, but with sudo. Not much different but just in case:

```
sudo lsusb -d 1b1c:0c13 -v                                                                                                    ✔ 

Bus 001 Device 005: ID 1b1c:0c13 Corsair H115i Platinum
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c13 H115i Platinum
  bcdDevice            1.00
  iManufacturer           1 Corsair
  iProduct                2 H115i Platinum
  iSerial                 3 7289_2.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower               64mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0001
  Self Powered
```

jurkovic-nikola, 2025-04-08:
Managed to source 360 version of the device, I'll see what can be done with it

jurkovic-nikola, 2025-04-09:
Try with latest 0.5.4 version.

tchubaba, 2025-04-10:
@jurkovic-nikola I tried it and it works great! Fantastic job! Thank you so much!