jurkovic-nikola/OpenLinkHub #95: RGB memomry cannot be managed despite proper configuration

IOTech17, 2025-03-23:
Hi,

I have followed up the guide on how to configurethe system to maganed the RGB memory via Openlinkhub, but the memory do not appear in the dashboard.

OS : Solus OS 4.7 budgie
MB : Asus Rog Strix X399-E Gaming
CPU : Threadripper 2950X
RAM : Corsair Vengeance RGB PRO SL DDR4 3200 MHZ (CMH128GX4M4E3200C16)
https://www.corsair.com/us/fr/p/memory/cmh128gx4m4e3200c16/vengeance-rgb-pro-sl-128gb-4x32gb-ddr4-dram-3200mhz-c16-memory-kit-a-black-cmh128gx4m4e3200c16

config.json
{
    "amdGpuIndex": 0,
    "cpuSensorChip": "",
    "debug": false,
    "decodeMemorySku": true,
    "enhancementKits": "",
    "exclude": [],
    "frontend": true,
    "listenAddress": "127.0.0.1",
    "listenPort": 27003,
    "logFile": "",
    "logLevel": "error",
    "manual": false,
    "memory": true,
    "memorySku": "CMH128GX4M4E3200C16",
    "memorySmBus": "i2c-10",
    "memoryType": 4,
    "metrics": false,
    "resumeDelay": 15000,
    "temperatureOffset": -27
}

this what I get if I run i2cdetect on the i2c index related to the memory

sudo i2cdetect -y 10
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- 19 -- 1b -- 1d -- 1f 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 31 -- -- 34 35 -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- 51 -- 53 -- 55 -- 57 -- 59 -- 5b -- 5d -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 

The motherboard have 8 dimm, I am using A1-B1 and C1-D1 for proper quad channel configuration

As you can see above everyhting is detected properly and ee1004 has been unloaded and blacklisted

I also have updated the udev rules

crw-------  1 openlinkhub root     89,  10 23 mars  07:42 i2c-10

The log file do not give me any error message but I still cannot the memory inside the dashboard.

Any idea what I am missing?

jurkovic-nikola, 2025-03-23:
If you set `memorySku` manually, than `decodeMemorySku` needs to be `false`

IOTech17, 2025-03-23:
I removed the part number and let decodeMemorySku to true. Restarted Openlinkhub but the issue remains

jurkovic-nikola, 2025-03-23:
Put `memorySku` back and set `decodeMemorySku` to false. Also, how many DIMMs you have populated ?. It seams 1 register  address is missing if you have 4 DIMMs installed.

IOTech17, 2025-03-23:
Interesting when looking for the part number all the 4 DIMMs are recognized

sudo dmidecode -t memory | grep 'Part Number'
	Part Number: CMH128GX4M4E3200C16
	Part Number: CMH128GX4M4E3200C16
	Part Number: CMH128GX4M4E3200C16
	Part Number: CMH128GX4M4E3200C16

I changed the config.json

{
    "amdGpuIndex": 0,
    "cpuSensorChip": "",
    "debug": false,
    "decodeMemorySku": false,
    "enhancementKits": "",
    "exclude": [],
    "frontend": true,
    "listenAddress": "127.0.0.1",
    "listenPort": 27003,
    "logFile": "",
    "logLevel": "error",
    "manual": false,
    "memory": true,
    "memorySku": "CMH128GX4M4E3200C16",
    "memorySmBus": "i2c-10",
    "memoryType": 4,
    "metrics": false,
    "resumeDelay": 15000,
    "temperatureOffset": -27
}

The issue remains

sudo i2cdetect -y 10
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- 19 -- 1b -- 1d -- 1f 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 31 -- -- 34 35 -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- 51 -- 53 -- 55 -- 57 -- 59 -- 5b -- -- -- 5f 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

I do not know why it doesn't show one of the registered addresses in I2C but the 128 GB is showing and everyhting is working fine.

jurkovic-nikola, 2025-03-23:
Dump stdout.log file here

IOTech17, 2025-03-23:
Please find the stdou.log output

cat /opt/OpenLinkHub/stdout.log
{"error":"exec: \"amd-smi\": executable file not found in $PATH","level":"error","msg":"Unable to process amd-smi","time":"2025-03-22T23:56:07+01:00"}
{"error":"exec: \"amd-smi\": executable file not found in $PATH","level":"error","msg":"Unable to process amd-smi","time":"2025-03-22T23:58:43+01:00"}
{"error":"exec: \"amd-smi\": executable file not found in $PATH","level":"error","msg":"Unable to process amd-smi","time":"2025-03-23T00:04:26+01:00"}
{"error":"exec: \"amd-smi\": executable file not found in $PATH","level":"error","msg":"Unable to process amd-smi","time":"2025-03-23T00:12:23+01:00"}
{"error":"exec: \"amd-smi\": executable file not found in $PATH","level":"error","msg":"Unable to process amd-smi","time":"2025-03-23T07:42:56+01:00"}
{"error":"exec: \"amd-smi\": executable file not found in $PATH","level":"error","msg":"Unable to process amd-smi","time":"2025-03-23T09:41:52+01:00"}
{"error":"exec: \"amd-smi\": executable file not found in $PATH","level":"error","msg":"Unable to process amd-smi","time":"2025-03-23T09:42:48+01:00"}
{"error":"exec: \"amd-smi\": executable file not found in $PATH","level":"error","msg":"Unable to process amd-smi","time":"2025-03-23T10:54:24+01:00"}

This is the only error I have (because I am using the opensource amd drivers)

IOTech17, 2025-03-23:
now it is showing all the registered address 

sudo i2cdetect -y 10
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- 19 -- 1b -- 1d -- 1f 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 31 -- -- 34 35 -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- 51 -- 53 -- 55 -- 57 -- 59 -- 5b -- 5d -- 5f 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

jurkovic-nikola, 2025-03-23:
Include the whole stdout.log file. Drag and drop here.

IOTech17, 2025-03-23:
[stdout.log](https://github.com/user-attachments/files/19408402/stdout.log)

jurkovic-nikola, 2025-03-23:
Change `"logLevel": "info"`, restart service and include file again.

IOTech17, 2025-03-23:
@jurkovic-nikola 

Please find the new log

[stdout.log](https://github.com/user-attachments/files/19409187/stdout.log)

IOTech17, 2025-03-23:
start","serial":"i2c10","time":"2025-03-23T14:21:06+01:00"}
{"level":"info","msg":"No such register found. Skipping...","register":80,"time":"2025-03-23T14:21:06+01:00"}
{"level":"info","msg":"No such register found. Skipping...","register":82,"time":"2025-03-23T14:21:06+01:00"}
{"level":"info","msg":"No such register found. Skipping...","register":84,"time":"2025-03-23T14:21:06+01:00"}
{"level":"info","msg":"No such register found. Skipping...","register":86,"time":"2025-03-23T14:21:06+01:00"}

jurkovic-nikola, 2025-03-23:
Thanks. Replace the binary with attached one and test. 

[OpenLinkHub.tar.gz](https://github.com/user-attachments/files/19409430/OpenLinkHub.tar.gz)

IOTech17, 2025-03-23:
After replacing the binary with the one you provided me, it works, I can see the memory tab in the dashboard and changed each DIMM RGB color.

jurkovic-nikola, 2025-03-23:
I'll push the fix in 0.5.3 out. Issue was parsing memory size, since I never took into a consideration 128G value of kit, so size was never parsed properly and DIMM was excluded.

IOTech17, 2025-03-23:
I though also that it was caused by the fact that I had 128GB of ram and that it was not recognized properly. Thank you for your help!

closed 2025-03-23 by jurkovic-nikola

IOTech17, 2025-03-23:
I can see new error now, everyhting is working fine though. 

I provide the new stdout.log just in case