jurkovic-nikola/OpenLinkHub #97: Corsair Scimitar Elite Wireless not showing up

KeyonAshton, 2025-03-28:
**Describe the solution you'd like**
Corsair Scimitar Elite Wireless MMO Gaming Mouse - Black & Yellow not showing up in the devices list.

**Additional context**
According to the supported devices, the Scimitar RGB Elite wired and wireless versions are supported, but not the version I use.
I would post my stdout.log but it is empty. I am still fairly new to Linux, and not really sure how I would get anything to populate the log file there.

[image]

**Desktop (please complete the following information):**
    OS: Ubuntu 24.10
    Asus ROG Strix B550-f
    Ryzen 5 5600x
    Corsair Scimitar Elite Wireless MMO Gaming Mouse - Black & Yellow

jurkovic-nikola, 2025-03-28:
Hello,

Please provide output of `lsusb -d 1b1c:` and link to Corsair website for your mouse.

KeyonAshton, 2025-03-28:
Bus 001 Device 005: ID 1b1c:0c1c Corsair CORSAIR iCUE Commander CORE
Bus 003 Device 013: ID 1b1c:2b22 Corsair CORSAIR SCIMITAR ELITE WIRELESS SE Gaming Mouse

[Corsair Scimitar Elite Wireless MMO Gaming Mouse - Black & Yellow](https://www.corsair.com/us/en/p/gaming-mouse/ch-9314415-ww/scimitar-elite-wireless-se-mmo-gaming-mouse-black-yellow-ch-9314415-ww)

KeyonAshton, 2025-03-30:
Forgot to say thank you for the quick response

jurkovic-nikola, 2025-04-23:
Device finally arrived, support will be pushed in 0.5.5 version.

closed 2025-05-18 by jurkovic-nikola