jurkovic-nikola/OpenLinkHub #98: unable to use any other layout except the US one

Domenicopezzullo, 2025-04-05:
**Describe the bug**
i cannot use another layout other than the US one, resulting in the ISO-only keys not lighting up

**Device / program info**
1. Content of `/opt/OpenLinkHub/config.json`
```
{
    "debug": false,
    "listenPort": 27003,
    "listenAddress": "127.0.0.1",
    "cpuSensorChip": "",
    "manual": false,
    "frontend": true,
    "metrics": false,
    "memory": false,
    "memorySmBus": "i2c-0",
    "memoryType": 4,
    "exclude": [],
    "decodeMemorySku": true,
    "memorySku": "",
    "resumeDelay": 15000,
    "logFile": "",
    "logLevel": "info",

[stdout.log](https://github.com/user-attachments/files/19617881/stdout.log)

    "enhancementKits": "",
    "temperatureOffset": 0,
    "amdGpuIndex": 0
}
```

3. Output of `lsusb` command
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 1532:0550 Razer USA, Ltd Razer Barracuda X 2.4
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 005: ID 1532:008c Razer USA, Ltd Razer DeathAdder V2 Mini
Bus 001 Device 006: ID 048d:5702 Integrated Technology Express, Inc. RGB LED Controller
Bus 001 Device 007: ID 03f0:f42a HP, Inc HP Neverstop Laser MFP 120x
Bus 001 Device 009: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 013: ID 1b1c:2b01 Corsair CORSAIR K70 CORE RGB TKL Mechanical Gaming Keyboard
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

5. Attach `/opt/OpenLinkHub/stdout.log` file.
too big so i'll just link  #it

**Screenshots**
None

**Desktop (please complete the following information):**
 - OS: CachyOS
 - 1.1.15
 - 0.5.2

**Additional context**
Add any other context about the problem here.

Domenicopezzullo, 2025-04-05:
oh also, i have a programmable button on my keyboard but have no way to program it

jurkovic-nikola, 2025-04-05:
Because there is only US layout. Other layouts are community effort to translate to something else.

closed 2025-04-05 by jurkovic-nikola