jurkovic-nikola/OpenLinkHub #100: [Features Request] Nexus - Locked Screen Mode & Additional Ideas

CrimsonKnight13, 2025-04-08:
I greatly appreciate that Nexus now has support! It's been a long while since I was able to use it after migrating from Windows. Here are my ideas & questions I have regarding the current implementation.

1.  When the system is locked, seeing default idle screensaver or the ability to utilize user-selected videos or images would be a nice QoL feature.
2. At this time, I'm unable to save a profile for the Nexus that shows anything but the CPU stats. If possible, could this be fixed? Additionally, real time changes to the item selection doesn't work.
3. Allow adjustment or creation of what stats are shown for single or multiple sources.
4. Does OLH recognize the Nexus' touchscreen capabilities? If so, can it be used to swipe between different info sources?

jurkovic-nikola, 2025-04-08:
1. I would like to avoid such integration to monitor your session state
2. There are multiple modes, CPU, GPU, Media Control and Dynamic Mode that displays AIO data if you have AIO in the system. 
3. Touch screen is supported for Media Control, where you can Volume Up / Down, Play / Pause, Stop, etc...

[image]

CrimsonKnight13, 2025-04-08:
1. For privacy or simplicity, I can understand that
2. Thanks, I tried the selections again & let the change occur. This did change the screen w/o need for a custom profile. Only snag I hit is with GPU statistics with a Radeon RX 7900 XT, which is doesn't change to it at all.
3. Thanks for the tip. It definitely works.

jurkovic-nikola, 2025-04-08:
For any AMD card, you will need `amd-smi` library installed on your system. This is core lib where GPU name, temps and load is pulled out. Without that lib, most of AMD card details won't work, same goes for NVIDIA, where `nvidia-smi` is used. `amd-smi` will work with native kernel drivers.

CrimsonKnight13, 2025-04-09:
Does OLH look for amdsmi or amd-smi? Arch Linux installs it as amd-smi (to /opt/rocm/bin) with no GPU output on Nexus.

jurkovic-nikola, 2025-04-09:
It's `amd-smi`. Check stdout.log if there are any errors regarding that command.

CrimsonKnight13, 2025-04-09:
I found that openlinkhub account wasn't seeing where amd-smi was located, so I used a systemd environment variable as an override.

Maybe this could be a configuration option to specify where amd-smi is located in a future release?

Thanks for the help & explanations!

jurkovic-nikola, 2025-04-12:
Hm, I've built this part of the code on Arch and never had to add anything on the system to get this command recognized. After the reboot, command was available.

CrimsonKnight13, 2025-04-12:
I'm running CachyOS & it appears that amdsmi installs to /opt/rocm/ for the optimized & vanilla packages. From the PKGBUILD, it doesn't appear to add /opt/rocm/bin to PATH for the system or any shell.

https://gitlab.archlinux.org/archlinux/packaging/packages/amdsmi/-/blob/main/PKGBUILD?ref_type=heads