jurkovic-nikola/OpenLinkHub #102: OpenRGB - Allow letting OpenRGB take care of LEDs

Doridian, 2025-04-13:
Pretty much add the OpenRGB protocol as a "client", so that an OpenRGB instance can connect to OpenLinkHub and I can manage all my RGB in one application.

This would also ideally coincide with full LED control (as in, each individual LED, not just setting a fixed effect per device, akin to OpenRGB's "direct" mode)

jurkovic-nikola, 2025-04-13:
TCP server is not planned at all. Those 2 products are not compatible since they manage the same thing and 2 programs cant communicate with the same device.

Doridian, 2025-04-13:
@jurkovic-nikola Hence asking for an  OpenRGB server. OpenRGB can connect to other instances speaking its protocol, so then I could have OpenRGB talk to OpenLinkHub and have one GUI to manage all devices.

Right now I have to use both, as all my other RGB stuff of course can't work in OpenLinkHub (since it is for Corsair only).
They both also work perfectly, they just can't talk to each other, which is the problem I'd like fixed.

jurkovic-nikola, 2025-04-13:
I do understand this, hence this was built because of same reason. Problem starts when I can't find proper documentation about OpenRGB network packet stack and how its handled. There are examples, but those are client examples, not server ones. Longest I got before with this, is to actually set protocol version. Whatever I tried to do more with this, openrgb process would just seg fault, and I would really want to avoid another reverse engineer challenge, since I had enough of this with Corsair protocol.

Doridian, 2025-04-13:
@jurkovic-nikola In that case, one last question: If I were to manage to get a working implementation of the OpenRGB protocol into OpenLinkHub, should I keep that in my own fork, or would you want/accept a PR of such feature?

jurkovic-nikola, 2025-04-14:
I don't mind a PR or maintenance at all, and thank you for your effort. https://github.com/jurkovic-nikola/OpenLinkHub/issues/103 will be implemented soon, so this will be much easier for you. Things to consider:

1. OpenLinkHub works only in device software mode. 
2. RGB data is updated from 10-40 ms. 
3. Each device has function `setDeviceColor` that takes care or everything, packet length, device position, byte buffer etc...
4. `setDeviceColor` will be updated to include https://github.com/jurkovic-nikola/OpenLinkHub/issues/103
5. When https://github.com/jurkovic-nikola/OpenLinkHub/issues/103 is complete, literally any RGB effect can be created.

NightSprinter, 2025-05-29:
I can understand the logic in this, honestly.  Given the original iCUE software also supports connecting to Philips HUE and Govee via their connecting desktop apps, and OpenRGB has some manner of support for those, it would make it easier to do so if someone wants to control their PC lighting and room lighting.

closed 2025-08-10 by jurkovic-nikola