jurkovic-nikola/OpenLinkHub #107: RAM temperature

Niram7777, 2025-04-18:
**Describe the solution you'd like**
Hello,
First *thank you* for this amazing project. You litteraly saved my computer because I dont have anything connected to the motherboard except iCUE/USB which is not raising my pump/fans although my liquid hits 60°C, CPU 90° !!!

1 - I managed to get RGB control over DRAM after some difficulties but to get temperatures I need spd5118 to setup temperature sensors. (But still remove it as specified in your doc)

```
# Steps needed to get RAM RGB and Temperature
[marin@pc-linux openlinkhub]$ sudo systemctl stop openlinkhub
[marin@pc-linux openlinkhub]$ sudo modprobe spd5118
[marin@pc-linux openlinkhub]$ sudo systemctl start openlinkhub
[marin@pc-linux openlinkhub]$ sudo rmmod spd5118
[marin@pc-linux openlinkhub]$ sudo systemctl restart openlinkhub
```
**Additional context**
motherboard: Aorus X870E Ice Pro
OpenLinkHub Build: 98715a69b66404189b454881177e0e6839cd0024
```
[    0.000000] Linux version 6.14.2-arch1-1 (linux@archlinux) (gcc (GCC) 14.2.1 20250207, GNU ld (GNU Binutils) 2.44) #1 SMP PREEMPT_DYNAMIC Thu, 10 Apr 2025 18:43:59 +0000
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-linux root=UUID=1a5e3457-0642-4f33-8f3d-cb1fc115dad2 rw loglevel=4 acpi_enforce_resources=lax
...
[  699.922966] spd5118 8-0051: DDR5 temperature sensor: vendor 0x00:0xb3 revision 2.2
[  699.930733] spd5118 8-0053: DDR5 temperature sensor: vendor 0x00:0xb3 revision 2.2
```
Do we have to implement some spd5118 code in OpenLinkHub or do we have to start/stop this module in your app or Linux service before starting ?

2 - random notes:
The configuration I set in the UI is getting lost.
In particular my XD7 LEDs are off all the time, I have to set 3-PIN Corsair RGB HUB to "QL RGB Series Fan" and "1 Device".
It apparently works if I modify the database/profiles/*json directly with those values for example
```
    "ExternalHubDeviceType": 5,
    "ExternalHubDeviceAmount": 1
```

Niram7777, 2025-04-18:
[image]
RAM temperature sensors need a hack with SPD5118
[image]

rainbow ❤️

jurkovic-nikola, 2025-04-18:
Hello,

1. If you want to use OpenLinkHub for RAM, you need to blacklist module, simple unload will just unload it for current session until machine reboots. 

2. From your `lsusb` output, its unclear where XD7 is connected, so it's hard to say where the problem is. Only CC and CCXT support external devices. 

Regarding MM700, its supported and working. You can check Supported Devices, if you device is not on the list, its not supported. Slipstream received will never show on the UI, just device connected to it if its supported. It's unclear what device you have paired.

Niram7777, 2025-04-18:
1. yes it was blacklisted but if I do this. I dont get temperature anymore (I still get RGB), this is why I am asking if there is a way to keep temperature monitoring?
2. XD7 is connected through MOLEX for the power and for the pump and RGB it is connected to:

> Bus 001 Device 004: ID 1b1c:0c2a Corsair CORSAIR iCUE COMMANDER CORE XT

So I guess there is no way to detect it is a XD7? (maybe check on Windows Corsair iCUE software it is able to autodetect what it is? -I am guessing no since there are no USB for XD7).
It is not a big issue since I have the RGB effect on the XD7 that I can control. just I have to hardcode it in the json or specify it manually each time.

3. `MM700 RGB Extended 3XL Cloth Gaming Mouse Pad / Desk Mat`

4. M65 WL White ,what is funny is that amoung all my devices, OpenRGB was only able to detect or control the RGB of this mouse

5. It was even worst before because I did not had 90° and 45° pipes, so it was one tube doing 180° by itself. I will either provide more length to the SSD-CPU tube or connect the SSD separetely to the XD7 (with the future GPU).

**Thanks for all your feedbacks, I dont know however when I can investigate it deeply.**

jurkovic-nikola, 2025-04-19:
1. Your RAM temperature is read by OpenLinkHub and can be used to modify fan speed. If you want to modify fan speed based on temperature, go to Profile and create new profile, select sensor Temperature Probe and your DIMM will be there. Program was built in such way it handles everything related to RAM from the code, without any external programs. 

2. You need to use option CUSTOM ARGB DEVICE, since 3 PIN CORSAIR RGB HUB is used for Lighting Node hub. This works and persist with restarts and profile is saved. Automatic detection of RGB devices is supported on all hubs, but this requires a 4th wire that is used for detection. This is so called sense wire and its used to transfer data about how much LEDs are on the device and to chain other devices. Automatic detection on ARGB 3-Pin is not possible, since you only have 3 wires, power line and data line to transfer RGB data. 

[image]

3. Is it smaller pad or larger ? https://www.corsair.com/us/en/p/mousepads/ch-9417070-ww/mm700-rgb-extended-mouse-pad-ch-9417070-ww or https://www.corsair.com/us/en/p/mousepads/ch-9417080-ww/mm700-rgb-extended-3xl-cloth-gaming-mouse-pad-desk-mat-ch-9417080-ww ?

4. Please send me link to Corsair site. Is it this one ? https://www.corsair.com/eu/en/p/gaming-mouse/ch-9319511-eu2/m65-rgb-ultra-wireless-tunable-fps-gaming-mousewhite-eu-ch-9319511-eu2

Niram7777, 2025-04-26:
1. Hmm, it is strange then that I dont get RAM temperature, maybe a permission issue on my side, I will investigate deeper.
2. Thanks for the explanation, the XD7 RGB connector is only 3-pin and connected to the unique "RGB" connector of the COMMANDER CORE XT (bottom right, next to "TEMP 2")
https://www.amazon.de/dp/B09GHTMJMY/ref=pe_27091401_487027711_TE_SCE_dp_i2
as specified in the Corsair doc
[image]

FYI I am not using the ARGB to motherboard adapter

[image]

3. larger https://www.corsair.com/us/en/p/mousepads/ch-9417080-ww/mm700-rgb-extended-3xl-cloth-gaming-mouse-pad-desk-mat-ch-9417080-ww
4. yes this one https://www.corsair.com/eu/en/p/gaming-mouse/ch-9319511-eu2/m65-rgb-ultra-wireless-tunable-fps-gaming-mousewhite-eu-ch-9319511-eu2

jurkovic-nikola, 2025-04-27:
1. If its permission issue, RAM wouldn't even show up. 
2. Added HYDRO X lineup RGB data for 0.5.5 version. You will have option to specify XD7 or any other HYDRO X component. 
3. Added for 0.5.5. 
4. Should arrive next week and will be released in 0.5.5.

jurkovic-nikola, 2025-05-03:
Devices implemented in 0.5.5.

Niram7777, 2025-05-05:
Thanks, looks better, I will test and investigate next release with M65 Wireless support, I hope I will have more time

[image]

jurkovic-nikola, 2025-05-05:
M65 was added in 0.5.5.

Niram7777, 2025-05-08:
https://aur.archlinux.org/packages/openlinkhub
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=openlinkhub
> [marin@pc-linux ~]$ sudo i2cdetect -l
> i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
> i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
> i2c-2	i2c       	AMDGPU DM i2c hw bus 0          	I2C adapter
> i2c-3	i2c       	AMDGPU DM i2c hw bus 1          	I2C adapter
> i2c-4	i2c       	AMDGPU DM i2c hw bus 2          	I2C adapter
> i2c-5	i2c       	AMDGPU DM i2c hw bus 3          	I2C adapter
> i2c-6	i2c       	AMDGPU DM aux hw bus 1          	I2C adapter
> i2c-7	i2c       	AMDGPU DM aux hw bus 2          	I2C adapter
> i2c-8	smbus     	SMBus PIIX4 adapter port 0 at 0b00	SMBus adapter
> i2c-9	smbus     	SMBus PIIX4 adapter port 2 at 0b00	SMBus adapter
> i2c-10	smbus     	SMBus PIIX4 adapter port 1 at 0b20	SMBus adapter

> [marin@pc-linux ~]$ sudo i2cdetect -y 8
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:                         -- -- -- -- -- -- -- -- 
> 10: -- -- -- -- -- -- -- -- -- 19 -- 1b -- -- -- -- 
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 40: -- -- -- -- -- -- -- -- -- 49 -- 4b -- -- -- -- 
> 50: -- 51 -- 53 -- -- -- -- -- -- -- -- -- -- -- -- 
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 70: -- -- -- -- -- -- -- --

> [marin@pc-linux ~]$ grep i2c /etc/udev/rules.d/98-corsair-memory.rules
> KERNEL=="i2c-8", MODE="0600", OWNER="openlinkhub"

> [marin@pc-linux ~]$ lsmod | grep -E "spd5118|i2c"
> i2c_piix4              36864  1
> i2c_smbus              20480  1 i2c_piix4
> i2c_dev                28672  2
> i2c_algo_bit           20480  1 amdgpu

> { //config.json
>     "amdGpuIndex": 0,
>     "amdsmiPath": "",
>     "checkDevicePermission": true,
>     "cpuSensorChip": "",
>     "cpuTempFile": "",
>     "debug": false,
>     "decodeMemorySku": false,
>     "enhancementKits": "",
>     "exclude": [],
>     "frontend": true,
>     "listenAddress": "0.0.0.0",
>     "listenPort": 27003,
>     "logFile": "",
>     "logLevel": "info",
>     "manual": false,
>     "memory": true,
>     "memorySku": "CMP96GX5M2B6600C32",
>     "memorySmBus": "i2c-8",
>     "memoryType": 5,
>     "metrics": false,
>     "resumeDelay": 15000,
>     "temperatureOffset": 0
> }

closed 2025-06-07 by jurkovic-nikola