jurkovic-nikola/OpenLinkHub #108: K55 RGB PRO Gaming Keyboard no showing up.

IncZX, 2025-04-19:
The Supported Devices List shows the K55 CORE RGB but my K55 RGB PRO Gaming Keyboard doesn't show on the dashboard.

I did 'lsusb -d 1b1c:' and got this:
Bus 003 Device 002: ID 1b1c:1ba4 Corsair CORSAIR K55 RGB PRO Gaming Keyboard

This is the link to the official Corsair website:
https://www.corsair.com/ww/en/p/keyboards/ch-9226765-pt/k55-rgb-pro-gaming-keyboard-pt-ch-9226765-pt

Am I missing something or is this keyboard not supported?

If it's not, could you add it please?

Thank you.

jurkovic-nikola, 2025-04-20:
It will take a couple of days for the device to arrive.

IncZX, 2025-04-20:
Thank you for the quick response.

jurkovic-nikola, 2025-05-03:
Device is implemented in 0.5.5 release. US layout only.

IncZX, 2025-05-03:
Thank you. Had some trouble updating from the previous version but eventually I just removed the old version altogether and reinstalled from scratch. It worked and it's fine now.

If at a later date you could add the PT layout, it would be nice.

closed 2025-05-03 by IncZX