jurkovic-nikola/OpenLinkHub #109: Some Fans, Pump, & LCD Not Functioning as Expected

black-swan42, 2025-04-23:
**Describe the bug**
Thank you so much for this project. Absolute lifesaver. This issue is likely my fault but after looking through documentation and closed issues I haven't been able to resolve. 

I'm running into an issue where my RX RGB fans (3 on radiator, 3 intake) are working correctly and appear in dashboard but 4 RX MAX RGB (3x 120mm, 1x 140mm) are not. They are on the same side of the ICUE HUB which I suspect could be the issue. I was using an even split but moved the TITAN AIO 360mm pump onto it's own side to try and get recognition (didn't work). I am also not getting my LCD screen recognized in Dashboard. 

**Device / program info**
1. Content of `/opt/OpenLinkHub/config.json`
2. Output of `lsusb` command
3. Attach `/opt/OpenLinkHub/stdout.log` file.

[stdout.log](https://github.com/user-attachments/files/19860058/stdout.log)

(1)
   "debug": false,
    "listenPort": 27003,
    "listenAddress": "127.0.0.1",
    "cpuSensorChip": "k10temp",
    "manual": false,
    "frontend": true,
    "metrics": true,
    "memory": false,
    "memorySmBus": "i2c-0",
    "memoryType": 5,
    "exclude": [],
    "decodeMemorySku": true,
    "memorySku": "",
    "resumeDelay": 15000,
    "logFile": "",
    "logLevel": "info",
    "enhancementKits": [80,82],
    "temperatureOffset": 0,
    "amdGpuIndex": 0
(2) 
Bus 001 Device 002: ID 1b1c:0c3f Corsair iCUE LINK System Hub
Bus 003 Device 004: ID 1b1c:1c23 Corsair HX1200i Power Supply
Bus 003 Device 005: ID 1b1c:0c4e Corsair CORSAIR iCUE LINK AIO LCD Screen Module
(3)
Attached

**Expected behavior**
Expected behavior was for 10 fans, the AIO pump, and the LCD screen to be recognized with control in the dashboard. 

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: (257.5-2-arch)
 - Device Firmware
 - Program Version: 0.5.2

**Additional context**
Do you think you could support recognition of 9070XT for monitoring in the future? (This may be an AMD-SMI path issue on my part).

jurkovic-nikola, 2025-04-23:
Hello,

`{"firmware":"1.8.369","level":"info","msg":"This firmware can support only 14 devices.","product":"iCUE LINK System Hub","serial":"1B99CE2902F5735DA80439DEA218FA0D","time":"2025-04-14T22:13:33-05:00"}`

`{"devices":6,"level":"info","msg":"Device amount matches position amount.","positions":6,"time":"2025-04-14T22:13:52-05:00"}`

Please update your LINK System Hub firmware to latest version. You'll need iCUE for that. Regarding amd-smi, you'll need that library, but it was reported already that it's not recognized in PATH. This will be resolved in 0.5.5 with direct binary path option or you can define your amd-smi path in global .profile system file.

jurkovic-nikola, 2025-04-24:
Did you manage to get firmware updated and pump recognized ?

black-swan42, 2025-04-24:
Thanks for checking in! Was working on it last night but ran into some issues spinning up a windows VM. Will update as soon as Im able to update firmware.

Not sure if it matters but I'm using a USB splitter for the LCD and Power Supply. Power Supply is recognized but LCD isn't. Could be what you mentioned with the pump not being recognized but they are separate iCue links.

jurkovic-nikola, 2025-04-24:
Yes, LCD won't show up if pump isnt recognized.

black-swan42, 2025-04-30:
Switched out the iCUE hub for the one that came with the Titan (USB-C + Updated Firmware) and everything worked!

Only thing not showing up is my GPU and i'm 99.9% that's a PATH issue I just haven't resolved.

Really appreciate the help and this project @jurkovic-nikola !

For anyone else who stumbles on this who ran into the same trouble I did... Updating iCUE firmware through a VM is not advised and I never got it to work. I eventually connected the iCUE hub to an external laptop running windows. The iCUE hub version with USB-C can be powered that way. Micro USB requires a power source.

closed 2025-04-30 by black-swan42