jurkovic-nikola/OpenLinkHub #110: Commander Core XT Fan Control

btxshark1, 2025-04-25:
**Describe the bug**
tl;dr - 4-pin PWM fans attached to the commander are not controllable / don't show up. 

tl- 
I have a pair of pumps (pwm header, they are powered elsewhere) and a plain old 4-pin fan attached to the first three ports on the "fans" side of the commander that I would like to be able to control [assign profiles]. The LEDs and temp sensors are working perfectly but I don't see any way to control the fans. 

**Device / program info**
1. Content of `/opt/OpenLinkHub/config.json`
```
{
    "debug": false,
    "listenPort": 27003,
    "listenAddress": "127.0.0.1",
    "cpuSensorChip": "",
    "manual": false,
    "frontend": true,
    "metrics": false,
    "memory": false,
    "memorySmBus": "i2c-0",
    "memoryType": 4,
    "exclude": [],
    "decodeMemorySku": true,
    "memorySku": "",
    "resumeDelay": 15000,
    "logFile": "",
    "logLevel": "info",
    "enhancementKits": "",
    "temperatureOffset": 0,
    "amdGpuIndex": 0
}
```
2. Output of `lsusb` command
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 046d:c332 Logitech, Inc. G502 Proteus Spectrum Optical Mouse
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 006: ID 30be:100c Schiit Audio Schiit Modi+
Bus 001 Device 007: ID 0bda:b850 Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 008: ID 1b1c:0c2a Corsair CORSAIR iCUE COMMANDER CORE XT
Bus 001 Device 009: ID 1b1c:0c3f Corsair iCUE LINK System Hub
Bus 001 Device 010: ID 048d:5702 Integrated Technology Express, Inc. RGB LED Controller
Bus 001 Device 012: ID 1395:009a DSEA A/S GSP 370
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. USB3.2 Hub
Bus 002 Device 003: ID 05e3:0626 Genesys Logic, Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 002: ID 04d9:0348 Holtek Semiconductor, Inc. Keyboard
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```
3. Attach `/opt/OpenLinkHub/stdout.log` file.
[stdout.log](https://github.com/user-attachments/files/19912908/stdout.log)

**Expected behavior**
PWM fans attached to the commander fan hub ports should be controllable 

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS:  Fedora 42
 - Device Firmware: 1.4.62
 - Program Version: Build: 98715a69b66404189b454881177e0e6839cd0024 / 0.5.4

**Additional context**
Add any other context about the problem here.
My goal here is to use the temp sensors from the commander to control both the fans on the system hub, AND the pumps on the commander itself. 
Side note: The "HD RGB Series Fan" on the commander is actually the front panel RGB on the 9000D case.

jurkovic-nikola, 2025-04-25:
Hello,

Extract and install RPM, change config.json value debug to true and restart service. This will log packets from the device and attach new stdout.log file. 

[OpenLinkHub-0.5.5-1.x86_64.zip](https://github.com/user-attachments/files/19914360/OpenLinkHub-0.5.5-1.x86_64.zip)

btxshark1, 2025-04-25:
Build: e63be7e694de880e75dd6cf1d22a5186cbf385b4 / 0.5.5

config.json 
```
{
    "amdGpuIndex": 0,
    "amdsmiPath": "",
    "cpuSensorChip": "",
    "debug": true,
    "decodeMemorySku": true,
    "enhancementKits": "",
    "exclude": [],
    "frontend": true,
    "listenAddress": "127.0.0.1",
    "listenPort": 27003,
    "logFile": "",
    "logLevel": "info",
    "manual": false,
    "memory": false,
    "memorySku": "",
    "memorySmBus": "i2c-0",
    "memoryType": 4,
    "metrics": false,
    "resumeDelay": 15000,
    "temperatureOffset": 0
}
```

[stdout.log](https://github.com/user-attachments/files/19914643/stdout.log)

Probe 1 has gone missing now. Maybe this commander module is defective. 

[image]

jurkovic-nikola, 2025-04-25:
`{"amount":6,"data":"00 08 00 09 00 06 01 01 01 01 01 01 00 00 00...","level":"info","msg":"getDevices()","serial":"830630716a4884baee79f4261091005f","time":"2025-04-25T12:53:34-06:00"}` - Hub returned 6 PWM ports back. 

`{"level":"info","msg":"getDevices() - Port Status","port":0,"serial":"830630716a4884baee79f4261091005f","status":1,"time":"2025-04-25T12:53:34-06:00"}
{"level":"info","msg":"getDevices() - Port Status","port":1,"serial":"830630716a4884baee79f4261091005f","status":1,"time":"2025-04-25T12:53:34-06:00"}
{"level":"info","msg":"getDevices() - Port Status","port":2,"serial":"830630716a4884baee79f4261091005f","status":1,"time":"2025-04-25T12:53:34-06:00"}
{"level":"info","msg":"getDevices() - Port Status","port":3,"serial":"830630716a4884baee79f4261091005f","status":1,"time":"2025-04-25T12:53:34-06:00"}
{"level":"info","msg":"getDevices() - Port Status","port":4,"serial":"830630716a4884baee79f4261091005f","status":1,"time":"2025-04-25T12:53:34-06:00"}
{"level":"info","msg":"getDevices() - Port Status","port":5,"serial":"830630716a4884baee79f4261091005f","status":1,"time":"2025-04-25T12:53:34-06:00"}`

All 6 ports are returning status `1`. Status 1 is nothing is connected to that physical port. 

How is your RGB connected ? Did you use 3-PIN CORSAIR RGB HUB or CUSTOM ARGB DEVICE. 3-PIN CORSAIR RGB is only valid for Lighting Hub.

btxshark1, 2025-04-25:
The three pin "RGB" header is connected to the 9000D front panel (backlights for the ports and logo) 
"RGB Hub" 1,2,3 are connected to various strips in the reservoirs and waterblocks. 
All of that is working.

Fan port one and two are [these](https://shop.alphacool.com/en/shop/pumpen/vpp-d5/pum-alphacool-vpp-apex-pump)
three and four are [these](https://www.corsair.com/us/en/p/case-fans/co-9050044-ww/ml140-140mm-pwm-premium-magnetic-levitation-fan-twin-pack-co-9050044-ww)
[image]

They all spin slowly like half duty cycle

jurkovic-nikola, 2025-04-26:
For test, try to unplug 9000D front panel RGB wire and restart service to see if anything will change.

jurkovic-nikola, 2025-04-27:
Weird, CCXT should not behave like this, but OK, great it works now. All pumps are auto detected for AIOs and LINK system, so there is no manual way to define port as pump, idk if this is even possible in iCUE. If you have 2 separate loops, there's really no way to find common sensor to control both pumps with same profile.

btxshark1, 2025-04-27:
Fair game.
Thank you for all your help and for this awesome project!

closed 2025-04-27 by btxshark1

donkeybanana, 2025-06-14:
@btxshark1 I have the Commander XT and have experienced intermittent issues like you described. No hard reset steps would work. 

So I had a Windows VM, I would pass the USB device through to regain control using iCue and all would be good for a while. I stopped using the device due to this issue, but it may be of some help to you.

Now I'm looking forward to trying this project, and I'm hoping the workaround won't be needed in the future!🤞

btxshark1, 2025-06-18:
👍 Good to know they play nice with passthru... Even just to have a way to update firmware if needed down the road. 

This one has been working flawlessly after the initial finicky setup. 
I've since added a second system hub to handle more fans (4x 480rads with push/pull... things got out of hand 😆 ) and the CCXT now has 4 "fans" (really 2 pumps and 2 pwm intake fans) and 6 leds - 5 alphacool things with [this](https://ebay.us/m/6IFYqn) adapter and the front panel of the 9000D. Everything is in sync and the new profile curves in 5.7 work great across all three corsair controllers.