jurkovic-nikola/OpenLinkHub #111: Please add support for Corsair iCUE H150i ELITE LCD XT Display Liquid CPU Cooler

eliacap, 2025-05-02:
Please add support for
iCUE H150i ELITE LCD XT Display Liquid CPU Cooler

https://www.corsair.com/us/en/p/cpu-coolers/cw-9060075-ww/icue-h150i-elite-lcd-xt-display-liquid-cpu-cooler

jurkovic-nikola, 2025-05-02:
This device is already supported via Commander Core

eliacap, 2025-05-02:
There is something I am not understanding, in my Linux Mint 22.1 distribution I only see these usb, all leds are managed correctly, but the LCD display I have no way to set it.

lsusb |grep Corsair
Bus 001 Device 003: ID 1b1c:1c1e Corsair HX1000i Power Supply
Bus 003 Device 005: ID 1b1c:0c32 Corsair CORSAIR iCUE COMMANDER Core
Bus 003 Device 006: ID 1b1c:0c2a Corsair CORSAIR iCUE COMMANDER CORE XT

eliacap, 2025-05-02:
lsusb -d 1b1c:
Bus 001 Device 003: ID 1b1c:1c1e Corsair HX1000i Power Supply
Bus 003 Device 005: ID 1b1c:0c32 Corsair CORSAIR iCUE COMMANDER Core
Bus 003 Device 006: ID 1b1c:0c2a Corsair CORSAIR iCUE COMMANDER CORE XT

jurkovic-nikola, 2025-05-02:
Because there is no LCD connected in your system, as far as lsusb can see. You need to use Y cable from the package and connect LCD display to Y cable alongside Commander Core USB cable.

eliacap, 2025-05-02:
my iCUE H150i ELITE LCD XT is directly connected to the asus x670e motherboard with the cable provided.
I will do as you suggested and update you

jurkovic-nikola, 2025-05-02:
Your iCUE H150i ELITE LCD XT is connected via flat cable to Commander Core hub that handles fans and RGB. There is also a second cable from the pump head (USB 2.0 header) that's used for LCD data. 

Note: Please do not touch pump head while machine is running, this will lead to device issue and potential damage due to current fluctuation. Always power off the system. 

Your Commander Core and Commander Core XTs are connected via USB2.0 header to mobo. Only thing from iCUE H150i ELITE LCD XT that can go to mobo i PWM single wire, and its used for signaling the BIOS that pump is working.

eliacap, 2025-05-02:
You are absolutely right and I thank you (sorry but I had assembled the case some time ago and I did not remember the connections).
I checked the USB connections and I had connected the USB cable of the commander core badly, I connected it to the Y cable and it is recognized by the kernel:

lsusb |grep Cors
Bus 001 Device 003: ID 1b1c:1c1e Corsair HX1000i Power Supply
Bus 003 Device 005: ID 1b1c:0c32 Corsair CORSAIR iCUE COMMANDER Core
Bus 003 Device 006: ID 1b1c:0c39 Corsair LCD Cap for Elite Capellix coolers
Bus 003 Device 007: ID 1b1c:0c2a Corsair CORSAIR iCUE COMMANDER CORE XT
Now I see the device in OpenLinkHub:

H150i ELITE LCD XT

Thanks for your support

jurkovic-nikola, 2025-05-02:
You can change LCD modes on pump itself, small screen icon. There are couple of pre-sets that you can use and this list gets updated how new LCD modes are made. If you want to use git, images are stored in `/opt/OpenLinkHub/database/lcd/images/`. Manual copy if file is required and service restart. Your image will show up in the dropdown then. Background changes will be implemented in LCD editor in newer versions, so you don't have to bother with manual actions.

eliacap, 2025-05-02:
Fantastic!
Thanks again

closed 2025-05-02 by jurkovic-nikola