jurkovic-nikola/OpenLinkHub #113: Does not pick up H150i LCD

Dormant-Hero, 2025-05-03:
**Describe the bug**

Does not pick up H150i LCD

**Device / program info**
1. Content of `/opt/OpenLinkHub/config.json`
Immutable OS. Nothing in this location:
[image]

I did find  something in my home folder:
[image]

2. Output of `lsusb` command
[lbusb.txt](https://github.com/user-attachments/files/20025385/lbusb.txt)

 
3. Attach `/opt/OpenLinkHub/stdout.log` file.
I don't have this:
[image]

I do however have this here:
[image]

I have now also added my gif as everything seems to work based of my home folder rather than opt due to immutable system?

**Expected behavior**
I expected to be able to change gif but nothing worked.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Bazzite
 - Device Firmware: Software states it is 2.11.221
 - Program Version: 0.5.5

**Additional context**
Add any other context about the problem here.

Tried to follow the docs for the api stuff and haslcd came out false for everything here. Probably not helpful but I don't work in tech... Hopefully this post is not insanely stupid as finance is my background.

[lcdcheck.txt](https://github.com/user-attachments/files/20025407/lcdcheck.txt)

jurkovic-nikola, 2025-05-04:
Hello, for immutable distro like Bazzite, please follow instruction for 4. Immutable distributions (Bazzite OS, SteamOS, etc...). Process is little bit different, since on immutable, program is installed into your home directory. 

This process replaces default `openlinkhub` username in `99-openlinkhub.rules` with yours. 

`{"device":"LCD Cap for Elite Capellix coolers","interface":0,"level":"info","msg":"Processing device...","productId":3123,"serial":"1944214680003","time":"2025-05-04T00:03:05+01:00"}
{"device":"LCD Cap for Elite Capellix coolers","level":"warning","msg":"Invalid permissions","path":"/dev/hidraw10","productId":3123,"time":"2025-05-04T00:03:05+01:00"}`

This means 99-openlinkhub.rules was never copied to `/etc/udev/rules.d`.

Dormant-Hero, 2025-05-04:
HI Jurkovic,

Firstly a huge thank you for making the linux community greater and creating this software.

I did attempt to follow step 4 using the immutable.sh (whatever name it had) but I may have also used the non immutable version as I initially copied the release in the readme example terminal commands like an idiot which did not have the immutable sh.

I then did the process correctly noticing it could just be an old build and follow the guide but for the latest release where the immutable file now existed.

Below you can see I already had the file you mentioned in this folder:
[image]

The file already contained the below (github does not allow .rules type):
[rules.txt](https://github.com/user-attachments/files/20026564/rules.txt)

jurkovic-nikola, 2025-05-04:
Hello, it will be faster if you join to Discord https://discord.gg/mPHcasZRPy

Dormant-Hero, 2025-05-04:
Spoke to Nikola on discord and found I had a new LCD ID that the software was not aware of or something along these lines. Nikola added and problem resolved.

closed 2025-05-04 by Dormant-Hero