jurkovic-nikola/OpenLinkHub #115: Button remappings on mouse not saving

TheOldKingCole, 2025-05-06:
So after failing to get CKB-Next to register my Ironclaw RGB Wireless over either the Dongle or Bluetooth I tried out OpenLinkHub and it works perfectly in every regard except it won't save any remapping I do to any of the the side buttons. I discovered the issue when I was replaying Doom Eternal and wanted to map the option button to Q for weapon wheel access since I was using the other side buttons for other controls/dpi and the while the web app would say it saved the config the option button wouldn't actually activate the q key in or out of game

**Device / program info**
1. {
    "debug": false,
    "listenPort": 27003,
    "listenAddress": "127.0.0.1",
    "cpuSensorChip": "",
    "manual": false,
    "frontend": true,
    "metrics": false,
    "memory": false,
    "memorySmBus": "i2c-0",
    "memoryType": 4,
    "exclude": [],
    "decodeMemorySku": true,
    "memorySku": "",
    "resumeDelay": 15000,
    "logFile": "",
    "logLevel": "info",
    "enhancementKits": "",
    "temperatureOffset": 0,
    "amdGpuIndex": 0,
    "amdsmiPath": "",
    "checkDevicePermission": true,
    "cpuTempFile": ""
}
2. Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 2dc8:6013 8BitDo Ultimate 2
Bus 001 Device 005: ID 1532:0568 Razer USA, Ltd Razer Kraken V4 Pro
Bus 001 Device 006: ID 1532:0c08 Razer USA, Ltd Razer Firefly V2 Pro
Bus 001 Device 007: ID 1532:028d Razer USA, Ltd Razer BlackWidow V4 Pro
Bus 001 Device 008: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 009: ID 1b1c:1bdc Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver
Bus 001 Device 010: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 001 Device 011: ID 1532:054a Razer USA, Ltd Razer Leviathan V2 X
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

3. 

[stdout.log](https://github.com/user-attachments/files/20066647/stdout.log)

 - OS: Linux Mint 22.1
 - Firmware version 5.1.135
 - Version 0.5.5.

jurkovic-nikola, 2025-05-06:
Hey, join to Discord, it will be faster. https://discord.gg/mPHcasZRPy

TheOldKingCole, 2025-05-07:
So for whatever reason it suddenly started working so I'm going to keep an eye on it and if I don't run into this issue again this week I'll just close the bug report and assume it was just a one of error.

jurkovic-nikola, 2025-05-07:
stdout.log points to a lot of errors that are common with usage of another software in combination with this.

closed 2025-05-18 by jurkovic-nikola