jurkovic-nikola/OpenLinkHub #117: Unable to process amd-smi

yipadam, 2025-05-12:
I am getting 

{"error":"exit status 1","level":"error","msg":"Unable to process amd-smi","time":"2025-04-19T20:14:27+10:00"}

when running the command i could get the following

amd-smi static -g 0 --asic --json 
[
    {
        "gpu": 0,
        "asic": {
            "market_name": "Radeon RX 7900 XTX",
            "vendor_id": "0x1002",
            "vendor_name": "Advanced Micro Devices Inc. [AMD/ATI]",
            "subvendor_id": "0x----",
            "device_id": "0x----",
            "subsystem_id": "0x----,
            "rev_id": "0xc8",
            "asic_serial": "--------------------",
            "oam_id": "N/A",
            "num_compute_units": 96,
            "target_graphics_version": "gfx1100"
        }
    }
]

jurkovic-nikola, 2025-05-12:
```
$ sudo -u openlinkhub bash
bash-5.2$ amd-smi static -g 0 --asic --json
[
    {
        "gpu": 0,
        "asic": {
            "market_name": "Granite Ridge [Radeon Graphics]",
            "vendor_id": "0x1002",
            "vendor_name": "Advanced Micro Devices Inc. [AMD/ATI]",
            "subvendor_id": "0x1458",
            "device_id": "0x13c0",
            "subsystem_id": "0xd000",
            "rev_id": "0xc1",
            "asic_serial": "N/A",
            "oam_id": "N/A",
            "num_compute_units": 2,
            "target_graphics_version": "gfx1036"
        }
    }
]
bash-5.2$ whoami
openlinkhub
bash-5.2$ 
```

[image]

yipadam, 2025-05-13:
Thanks for the response!!
sudo usermod -a -G render,video openlinkhub
Fixed it, it was working when i first install openLinkhub not sure what happened.
Thanks!!!

closed 2025-05-13 by jurkovic-nikola

jurkovic-nikola, 2025-05-13:
What distro you're using ? This appears to be specific issue and maybe documentation should be updated to include this.