jurkovic-nikola/OpenLinkHub #122: Logs take all space available on device

alexey-mercari, 2025-05-21:
**Describe the bug**
I installed OpenLinkHub about a week ago, enabled it via systemd, and left it running.

Yesterday, I began encountering “No space left on device” errors. After cleaning up my disk, the issue temporarily disappeared but reoccurred today.

Upon further investigation, I discovered that /var/lib/openlinkhub/stdout.log had grown to over 1 TB in size, consuming nearly all available disk space.

```
$ ls -la /var/lib/openlinkhub/stdout.log
.rwxr-xr-x openlinkhub openlinkhub 1.8 TB Wed May 21 10:03:46 2025  /var/lib/openlinkhub/stdout.log
```

**Device / program info**
1. Content of `/opt/OpenLinkHub/config.json`
```
 $ cat /var/lib/openlinkhub/config.json
{
    "debug": false,
    "listenPort": 27003,
    "listenAddress": "127.0.0.1",
    "cpuSensorChip": "",
    "manual": false,
    "frontend": true,
    "metrics": false,
    "memory": false,
    "memorySmBus": "i2c-0",
    "memoryType": 4,
    "exclude": [],
    "decodeMemorySku": true,
    "memorySku": "",
    "resumeDelay": 15000,
    "logFile": "",
    "logLevel": "info",
    "enhancementKits": "",
    "temperatureOffset": 0,
    "amdGpuIndex": 0,
    "amdsmiPath": "",
    "checkDevicePermission": true,
    "cpuTempFile": ""

```
3. Output of `lsusb` command
```
#  lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 1b1c:1bdc Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 2109:2813 VIA Labs, Inc. VL813 Hub
Bus 003 Device 003: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub
Bus 003 Device 004: ID 2109:2813 VIA Labs, Inc. VL813 Hub
Bus 003 Device 005: ID 1532:00a4 Razer USA, Ltd Razer Mouse Dock Pro
Bus 003 Device 006: ID 046d:c539 Logitech, Inc. Lightspeed Receiver
Bus 003 Device 008: ID 08bb:2912 Texas Instruments PCM2912A Audio Codec
Bus 003 Device 009: ID 2104:0313 Tobii Technology AB EyeChip
Bus 003 Device 010: ID 0bda:5412 Realtek Semiconductor Corp. USB2.1 Hub
Bus 003 Device 011: ID 2109:8883 VIA Labs, Inc. USB Billboard Device
Bus 003 Device 012: ID 413c:d001 Dell Computer Corp. Dell Presence Sensor WB7022
Bus 003 Device 013: ID 0bda:1100 Realtek Semiconductor Corp. HID Device
Bus 003 Device 016: ID 1b1c:1bab Corsair CORSAIR K100 RGB AIR WIRELESS Ultra-Thin Mechanical Gaming Keyboard
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 002: ID 2109:0813 VIA Labs, Inc. VL813 Hub
Bus 004 Device 003: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub
Bus 004 Device 004: ID 2109:0813 VIA Labs, Inc. VL813 Hub
Bus 004 Device 005: ID 0bda:0412 Realtek Semiconductor Corp. USB3.2 Hub
Bus 004 Device 006: ID 413c:c015 Dell Computer Corp. Dell Webcam WB7022
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 005 Device 005: ID 0414:a000 Giga-Byte Technology Co., Ltd USB Audio
Bus 005 Device 006: ID 8087:0029 Intel Corp. AX200 Bluetooth
[... 87 lines cut ...]

DISTRIB_ID="Arch"
DISTRIB_RELEASE="rolling"
DISTRIB_DESCRIPTION="Arch Linux"
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://gitlab.archlinux.org/groups/archlinux/-/issues"
PRIVACY_POLICY_URL="https://terms.archlinux.org/docs/privacy-policy/"
LOGO=archlinux-logo
```
 - Device Firmware
n/a
 - Program Version [e.g. 0.3.9]
`0.5.5`

**Additional context**
Add any other context about the problem here.

Installed via aur: https://aur.archlinux.org/packages/openlinkhub

jurkovic-nikola, 2025-05-21:
Change your log level to error in config.json to reduce amount of logs

jurkovic-nikola, 2025-05-21:
Will be fixed in 0.5.7. This is related to rgb profile change and how device is connected. error level is just a temp fix.

alexey-mercari, 2025-05-22:
Thanks for the fast response!

closed 2025-06-07 by jurkovic-nikola