jurkovic-nikola/OpenLinkHub #125: Executable Temperature Source

isugimpy, 2025-05-28:
**Describe the solution you'd like**
As a user, I'd like to be able to execute arbitrary scripts or executables to provide a numeric output for a sensor, and use that output to determine fan speed.

**Additional context**
I feel like there could be many ways this could be used in general.  In my case, my liquid temperature in my AIO isn't detected by OpenLinkHub, leaving me only able to really use CPU temperature as a control for fan speed.  However, the CPU temperature is largely irrelevant, when the liquid temperature hasn't reached saturation.  Being able to parse the output of liquidctl, or some other arbitrary executable, would allow much more useful control.

jurkovic-nikola, 2025-07-05:
This will be released in 0.6.0. You'll have option to define binary path in profile creation. 

[image]

isugimpy, 2025-07-05:
Awesome!  Thank you for this, I'll look forward to trying it!

jurkovic-nikola, 2025-07-05:
Program will look for single output of binary, it can be float or int, it will parse.

jurkovic-nikola, 2025-07-06:
Released in 0.6.0

closed 2025-07-06 by jurkovic-nikola