jurkovic-nikola/OpenLinkHub #130: Tried to install it on Linux Mint.

TheLaughingThird, 2025-06-07:
I appreciate the effort to make this. I am on Linux mint. Today I tried to install this repo but cant get it to work. 

_As an alternative, OpenLinkHub can be run in Docker, using the Dockerfile in this repository to build it locally. A configuration file has to be mounted to /opt/OpenLinkHub/config.json_

1. Clone repo
2. Create **/opt/OpenLinkHub/config.json**   I created a FILE see enclosed don't know if this is correct, or should I create a folder ? 
3. Run these commands:

`docker build . -t openlinkhub`

`docker run --network host --privileged -v ./config.json:/opt/OpenLinkHub/config.json openlinkhub`

### OUTPUT:

`➜  OpenLinkHub git:(main) docker build . -t openlinkhub
[+] Building 1.1s (20/20) FINISHED                               docker:default
 => [internal] load build definition from Dockerfile                       0.0s
 => => transferring dockerfile: 917B                                       0.0s
 => [internal] load metadata for docker.io/library/debian:bullseye-slim    0.9s
 => [internal] load metadata for docker.io/library/golang:1.23.8-bullseye  0.9s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => [build 1/7] FROM docker.io/library/golang:1.23.8-bullseye@sha256:480b  0.0s
 => [stage-1 1/8] FROM docker.io/library/debian:bullseye-slim@sha256:2f23  0.0s
 => CACHED [stage-1 2/8] RUN apt-get update &&     apt-get install -y lib  0.0s
 => CACHED [stage-1 3/8] RUN mkdir -p /opt/OpenLinkHub                     0.0s
 => CACHED [build 2/7] RUN apt-get update && apt-get install -y libudev-d  0.0s
 => CACHED [build 3/7] WORKDIR /app                                        0.0s
 => CACHED [build 4/7] RUN git clone https://github.com/jurkovic-nikola/O  0.0s
 => CACHED [build 5/7] WORKDIR /app/OpenLinkHub                            0.0s
 => CACHED [build 6/7] RUN if [ -n "$GIT_TAG" ]; then git checkout "$GIT_  0.0s
 => CACHED [build 7/7] RUN go build .                                      0.0s
 => CACHED [stage-1 4/8] COPY --from=build /app/OpenLinkHub/OpenLinkHub /  0.0s
 => CACHED [stage-1 5/8] COPY --from=build /app/OpenLinkHub/database /opt  0.0s
 => CACHED [stage-1 6/8] COPY --from=build /app/OpenLinkHub/static /opt/O  0.0s
 => CACHED [stage-1 7/8] COPY --from=build /app/OpenLinkHub/web /opt/Open  0.0s
 => CACHED [stage-1 8/8] WORKDIR /opt/OpenLinkHub                          0.0s
 => exporting to image                                                     0.0s
 => => exporting layers                                                    0.0s
 => => writing image sha256:51979969adcf7cb5d82756c44314c7dda49547f7ff20c  0.0s
 => => naming to docker.io/library/openlinkhub                             0.0s
➜  OpenLinkHub git:(main) docker run --network host --privileged -v ./config.json:/opt/OpenLinkHub/config.json openlinkhub

panic: read /opt/OpenLinkHub/config.json: is a directory

goroutine 1 [running]:
OpenLinkHub/src/config.upgradeFile({0xc00002c0c0?, 0xc000014360?})
	/app/OpenLinkHub/src/config/config.go:129 +0x38a
OpenLinkHub/src/config.Init()
	/app/OpenLinkHub/src/config/config.go:72 +0xab
OpenLinkHub/src/controller.Start()
	/app/OpenLinkHub/src/controller/controller.go:27 +0x3f
main.main()
	/app/OpenLinkHub/main.go:26 +0x1f
➜  OpenLinkHub git:(main) `

jurkovic-nikola, 2025-06-07:
Hey, remove `-v ./config.json:/opt/OpenLinkHub/config.json`. This is from old builds where config.json wasnt generated by the program itself.

TheLaughingThird, 2025-06-07:
Hey, appriciate your work and reaction thank you!

Now it runs with command:

 `docker run --network host --privileged openlinkhub`

I really looking forward to now get some more silence in the room by setting ventilator speeds :)

Again, great job creating this thank you very much!

jurkovic-nikola, 2025-06-07:
You're welcome, readme is updated to reflect this change.

TheLaughingThird, 2025-06-07:
I got it to work by creating this **docker-compose.yaml** :

`services:
  openlinkhub:
    build: .
    privileged: true
    network_mode: host
    volumes:
      - ./config.json:/opt/OpenLinkHub/config.json
      - openlinkhub-data:/opt/OpenLinkHub/database
    restart: unless-stopped

volumes:
  openlinkhub-data:`

Also i created a **config.json** in the root:

`
{
  "debug": false,
  "listenPort": 27003,
  "listenAddress": "127.0.0.1",
  "cpuSensorChip": "k10temp",
  "manual": false,
  "frontend": true,
  "metrics": true,
  "resumeDelay": 15000,
  "memory": false,
  "memorySmBus": "i2c-0",
  "memoryType": 4,
  "exclude": [],
  "decodeMemorySku": true,
  "memorySku": "",
  "logFile": "",
  "logLevel": "info",
  "enhancementKits": [],
  "temperatureOffset": 0,
  "amdGpuIndex": 0,
  "amdsmiPath": "",
  "cpuTempFile": ""
}
`

Now as soon as i boot this container runs and you have this wonderfull dashboard ready on: [http://127.0.0.1:27003/](http://127.0.0.1:27003/)

jurkovic-nikola, 2025-06-07:
If you want, make a PR and include yaml so others can use it also.