jurkovic-nikola/OpenLinkHub #134: Press and Hold does not work for Mouse Forward bindings

Connicpu, 2025-06-09:
**Describe the bug**
With my Scimitar RGB Elite programmed to have one of its buttons trigger the "(Mouse) Forward" binding and "Press and Hold" checked, pressing the button only produces a momentary trigger regardless of how long I hold the button down for.

**Device / program info**
1. Content of `/opt/OpenLinkHub/config.json`
``` 
{
    "debug": false,
    "listenPort": 27003,
    "listenAddress": "127.0.0.1",
    "cpuSensorChip": "",
    "manual": false,
    "frontend": true,
    "metrics": false,
    "memory": true,
    "memorySmBus": "i2c-2",
    "memoryType": 5,
    "exclude": [],
    "decodeMemorySku": true,
    "memorySku": "",
    "resumeDelay": 15000,
    "logFile": "",
    "logLevel": "info",
    "enhancementKits": "",
    "temperatureOffset": 0,
    "amdGpuIndex": 0,
    "amdsmiPath": "",
    "checkDevicePermission": true,
    "graphProfiles": false,
    "cpuTempFile": ""
}
```
2. Output of `lsusb` command
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0572:1b08 Conexant Systems (Rockwell), Inc. Hi-Res Audio
Bus 003 Device 003: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 004: ID 174c:2174 ASMedia Technology Inc. ASMT2307
Bus 003 Device 005: ID 1b1c:1bfd Corsair CORSAIR K70 CORE RGB Mechanical Gaming Keyboard
Bus 003 Device 006: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 008: ID 0b05:1b9b ASUSTek Computer, Inc. USB Audio
Bus 003 Device 010: ID 045e:02fe Microsoft Corp. Xbox Wireless Adapter for Windows
Bus 003 Device 011: ID 1050:0406 Yubico.com Yubikey 4/5 U2F+CCID
Bus 003 Device 012: ID 0b05:19af ASUSTek Computer, Inc. AURA LED Controller
Bus 003 Device 013: ID 1b1c:0c41 Corsair H150i ELITE WH
Bus 003 Device 014: ID 13d3:3602 IMC Networks Wireless_Device
Bus 003 Device 015: ID 1b1c:1be3 Corsair CORSAIR SCIMITAR RGB ELITE Gaming Mouse
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 004 Device 003: ID 174c:3174 ASMedia Technology Inc. ASMT2307
Bus 004 Device 004: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 011 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 012 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```
3. Attach `/opt/OpenLinkHub/stdout.log` file.
```
{"file":"/var/lib/openlinkhub/dashboard.json","level":"info","message":"Dashboard file is missing, creating initial one.","timestamp":"2025-06-08T14:15:03-07:00"}
{"file":"/var/lib/openlinkhub/dashboard.json","level":"info","message":"Dashboard file is created.","timestamp":"2025-06-08T14:15:03-07:00"}
{"file":"/var/lib/openlinkhub/database/scheduler.json","level":"info","message":"Scheduler file is missing, creating initial one.","timestamp":"2025-06-08T14:15:03-07:00"}
{"file":"/var/lib/openlinkhub/database/scheduler.json","level":"info","message":"Scheduler file is created.","timestamp":"2025-06-08T14:15:03-07:00"}
{"error":{"Op":"open","Path":"/dev/uinput","Err":13},"level":"error","message":"Failed to open /dev/uinput","timestamp":"2025-06-08T14:15:05-07:00"}
{"error":{"Op":"open","Path":"/dev/uinput","Err":13},"level":"error","message":"Failed to create virtual keyboard","timestamp":"2025-06-08T14:15:05-07:00"}
[... 202 lines cut ...]
{"device":"CORSAIR SCIMITAR RGB ELITE Gaming Mouse","interface":3,"level":"info","message":"Processing device...","productId":7139,"serial":"1AB60DB1191B3F5BA4EC07CE3B0278A9","timestamp":"2025-06-08T20:42:29-07:00"}
{"device":"/dev/i2c-2","error":{"Op":"open","Path":"/dev/i2c-2","Err":13},"level":"error","message":"Unable to open i2c device","timestamp":"2025-06-08T20:42:29-07:00"}
{"error":{},"level":"warn","message":"No valid I2C devices found","timestamp":"2025-06-08T20:42:29-07:00"}
{"level":"info","message":"Starting USB monitor...","timestamp":"2025-06-08T20:42:29-07:00"}
{"level":"info","location":"/var/lib/openlinkhub/database/profiles/3137.json","message":"Loaded custom user profile","serial":"3137","timestamp":"2025-06-08T20:42:29-07:00"}
{"level":"info","location":"/var/lib/openlinkhub/database/profiles/1AB60DB1191B3F5BA4EC07CE3B0278A9.json","message":"Loaded custom user profile","serial":"1AB60DB1191B3F5BA4EC07CE3B0278A9","timestamp":"2025-06-08T20:42:29-07:00"}
{"level":"info","location":"/var/lib/openlinkhub/database/profiles/1AB60DB1191B3F5BA4EC07CE3B0278A9.json","message":"Loaded custom user profile","serial":"1AB60DB1191B3F5BA4EC07CE3B0278A9","timestamp":"2025-06-08T20:42:29-07:00"}
{"level":"info","location":"/var/lib/openlinkhub/database/profiles/3137.json","message":"Loaded custom user profile","serial":"3137","timestamp":"2025-06-08T20:42:29-07:00"}
{"level":"info","message":"Device successfully initialized","product":"SCIMITAR RGB ELITE","serial":"1AB60DB1191B3F5BA4EC07CE3B0278A9","timestamp":"2025-06-08T20:42:29-07:00"}
{"level":"info","message":"Device successfully initialized","product":"iCUE H150i RGB ELITE","serial":"3137","timestamp":"2025-06-08T20:42:29-07:00"}
{"level":"info","location":"/var/lib/openlinkhub/database/profiles/0300C00BAFDC1CA065567ECAF5001BC5.json","message":"Loaded custom user profile","serial":"0300C00BAFDC1CA065567ECAF5001BC5","timestamp":"2025-06-08T20:42:29-07:00"}
{"level":"info","location":"/var/lib/openlinkhub/database/profiles/0300C00BAFDC1CA065567ECAF5001BC5.json","message":"Loaded custom user profile","serial":"0300C00BAFDC1CA065567ECAF5001BC5","timestamp":"2025-06-08T20:42:29-07:00"}
{"level":"info","message":"Device successfully initialized","product":"K70 CORE RGB","serial":"0300C00BAFDC1CA065567ECAF5001BC5","timestamp":"2025-06-08T20:42:29-07:00"}
```

**Expected behavior**
I would expect when I hold the button a continuous pressing of that button is produced until it is released.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Arch Linux (KDE Plasma)
 - Device Firmware: 1.2.21
 - Program Version 0.5.6

jurkovic-nikola, 2025-06-09:
Press and Hold is ignored for Mouse actions, since this is not supported. You can not send rapid events for mouse operations in such fashion.

Connicpu, 2025-06-09:
Is there any way to have it so it just functions as a normal button and doesn't send the release until the mouse button has been released?

jurkovic-nikola, 2025-06-09:
I'll push it for 0.5.8 for this mouse, so you can test it. I'm still not sure how will this work, maybe your app can track if Forward has a long press or something.

Connicpu, 2025-06-10:
Couldn't get it to run properly on the latest branch, but I patched your scimitar.go/mouse.go changes onto 0.5.6 and it's working flawlessly thank you so much!

PR #135 (2025-06-10): feat(scimitar): support press-and-hold for all buttons and multi-button pressing
Fixes #134

Re-implements key assignment triggering for the scimitar by tracking the previously set bits and diffing them with the new value to determine which buttons are newly pressed down and which are newly released.

For press-and-hold key assignments, when they are initially pressed we generate a press and sync event same as before, but now we precisely track when the buttons are released and when that happens we only generate the key release and sync rather than an extra press, sync, release, sync.
commit 5e1bf2de (2025-06-10): feat(scimitar): support press-and-hold for all buttons and multi-button pressing

fixes #134