jurkovic-nikola/OpenLinkHub #136: Trying to figure out how to setup cpu temp based color mapping for the devices.

WicNaradha, 2025-06-11:
[image]

I set it up as such through the webUi, but it seems to not change based on the temperature at all. I also tried to call post  calls from a python script on the API but seems like that is not a feature yet to set the color of the device.

jurkovic-nikola, 2025-06-11:
You can check color values in RGB Editor for cpu temp profile. Color should change as cpu temp changes, but I'll check today.

jurkovic-nikola, 2025-06-11:
Color was calculated twice. Its fixed and will be available in next release.

WicNaradha, 2025-06-15:
Hey tested it out and yes it works now and is awesome. 
Would it be possible to have a kind of low pass filter on the sensor readings to not have it flicker and be a bit more smoother. Can I do a feature request?

jurkovic-nikola, 2025-06-15:
Will be added in 0.6.0

jurkovic-nikola, 2025-06-16:
Critical temp is program is set to 57C, in order to recover it needs to reach -5 delta, so 52.

WicNaradha, 2025-06-17:
I did some research actually... and it seems that I had triggered the an over current fault (in my firmware version was a fault positive so a bug I assume). It functioned normally on reset, and the coolant temp never exceeded 40, so is there a way to update the firmware without windows?