jurkovic-nikola/OpenLinkHub #139: Support for the Scimitar Elite

Niddrym, 2025-06-18:
Currently the only scimitar rgb elite model supported is 1be3

Could the 1b8b model be supported as well?

Output of lsusb 
Bus 009 Device 009: ID 1b1c:1b8b Corsair CORSAIR SCIMITAR RGB ELITE Gaming Mouse

jurkovic-nikola, 2025-06-18:
Send me some images of your mouse, its difficult to source it from currently active model.

Niddrym, 2025-06-18:
DPI min value 100 max 18000

[image]

Core buttons:

[image]

Side Buttons 1-12 (image only shows the 10)

[image]

The mappings are empty because I had to install iCue on a laptop since I'm running Linux on my main PC now.

jurkovic-nikola, 2025-06-18:
This looks like a identical device like current scimitar. I'll see what can be done.

jurkovic-nikola, 2025-06-22:
Can you join to Discord ?

jurkovic-nikola, 2025-06-22:
Tag me on general channel.

jurkovic-nikola, 2025-06-24:
Mouse is implemented and will be in next release. Thanks @Niddrym for testing.

closed 2025-06-29 by jurkovic-nikola