jurkovic-nikola/OpenLinkHub #141: Random crashing

depcustodian, 2025-06-22:
First of all, thank you so much for OpenLinkHub. I was so tired of using OpenRGB+liquidctl+custom GUI for configuring Corsair RGB and liquid cooling. A great solution right out of box!
**Describe the bug**
Still, i've encountered a random crash. The app is running well roght from the start of the system as intended, but then at a random time without any panic or error crashes. Restart won't help. Well, it do restart but wont initialize any Corsair device.
Also, i've experienced a rqandom problem with initializing of keyboard and mouse in OLH. It randomly either initialize them or no for no visible reason.
**Device / program info**
1. Content of `/opt/OpenLinkHub/config.json`
```
    "debug": false,
    "listenPort": 27003,
    "listenAddress": "127.0.0.1",
    "cpuSensorChip": "",
    "manual": false,
    "frontend": true,
    "metrics": false,
    "memory": false,
    "memorySmBus": "i2c-0",
    "memoryType": 4,
    "exclude": [],
    "decodeMemorySku": true,
    "memorySku": "",
    "resumeDelay": 15000,
    "logFile": "",
    "logLevel": "info",
    "enhancementKits": "",
    "temperatureOffset": 0,
    "amdGpuIndex": 0,
    "amdsmiPath": "",
    "checkDevicePermission": true,
    "graphProfiles": false,
    "cpuTempFile": "",
    "ramTempViaHwmon": false
```
3. Output of `lsusb` command
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0ecb:1f58 AKG C44-USB Microphone AKG C44-USB Microphone
Bus 001 Device 003: ID 1b1c:1bc6 Corsair CORSAIR K70 RGB PRO Optical-Mechanical Gaming Keyboard
Bus 001 Device 004: ID 1b1c:0a51 Corsair CORSAIR VOID ELITE Wireless Gaming Dongle
Bus 001 Device 005: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 006: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 007: ID 1b1c:1ba6 Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver
Bus 001 Device 008: ID 048d:5702 Integrated Technology Express, Inc. RGB LED Controller
Bus 001 Device 009: ID 1b1c:0c2a Corsair CORSAIR iCUE COMMANDER CORE XT
Bus 001 Device 010: ID 8087:0033 Intel Corp. AX211 Bluetooth
Bus 001 Device 011: ID 1b1c:0c2a Corsair CORSAIR iCUE COMMANDER CORE XT
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```
5. Attach `/opt/OpenLinkHub/stdout.log` file.
P.S. while making a log OLH exprerienced several crashes
```
device...","productId":3114,"serial":"d10c30310a4884baa3c214266091005f","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR iCUE COMMANDER CORE XT","interface":1,"level":"info","message":"Processing device...","productId":3114,"serial":"d10c30310a4884baa3c214266091005f","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR iCUE COMMANDER CORE XT","interface":0,"level":"info","message":"Processing device...","productId":3114,"serial":"c10c30010a4884bacb7d04261091005f","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR iCUE COMMANDER CORE XT","interface":1,"level":"info","message":"Processing device...","productId":3114,"serial":"c10c30010a4884bacb7d04261091005f","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR K70 RGB PRO Optical-Mechanical Gaming Keyboard","interface":0,"level":"info","message":"Processing device...","productId":7110,"serial":"F500190561C10828AB1885081802C009","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR K70 RGB PRO Optical-Mechanical Gaming Keyboard","interface":0,"level":"info","message":"Processing device...","productId":7110,"serial":"F500190561C10828AB1885081802C009","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR K70 RGB PRO Optical-Mechanical Gaming Keyboard","interface":0,"level":"info","message":"Processing device...","productId":7110,"serial":"F500190561C10828AB1885081802C009","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR K70 RGB PRO Optical-Mechanical Gaming Keyboard","interface":1,"level":"info","message":"Processing device...","productId":7110,"serial":"F500190561C10828AB1885081802C009","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR K70 RGB PRO Optical-Mechanical Gaming Keyboard","interface":2,"level":"info","message":"Processing device...","productId":7110,"serial":"F500190561C10828AB1885081802C009","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
{"device":"CORSAIR VOID ELITE Wireless Gaming Dongle","interface":3,"level":"info","message":"Processing device...","productId":2641,"serial":"","timestamp":"2025-06-22T18:17:47+03:00"}
[... 27 lines cut ...]
{"level":"info","message":"Device successfully initialized","product":"K70 PRO RGB","serial":"F500190561C10828AB1885081802C009","timestamp":"2025-06-22T18:17:48+03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/d10c30310a4884baa3c214266091005f.json","message":"Loaded custom user profile","serial":"d10c30310a4884baa3c214266091005f","timestamp":"2025-06-22T18:17:48+03:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/c10c30010a4884bacb7d04261091005f.json","message":"Loaded custom user profile","serial":"c10c30010a4884bacb7d04261091005f","timestamp":"2025-06-22T18:17:48+03:00"}
{"level":"info","message":"Device successfully initialized","product":"iCUE COMMANDER CORE XT","serial":"c10c30010a4884bacb7d04261091005f","timestamp":"2025-06-22T18:17:48+03:00"}
{"level":"info","message":"Device successfully initialized","product":"iCUE COMMANDER CORE XT","serial":"d10c30310a4884baa3c214266091005f","timestamp":"2025-06-22T18:17:48+03:00"}
```

**Expected behavior**
Obviously not to crash and succesfully initialize keyboard and mouse

**Desktop (please complete the following information):**
Chassis : Desktop  
OS : Arch Linux
Kernel : 6.15.2-zen1-1-zen
WM : Hyprland
CPU : Intel(R) Core(TM) i9-14900KF @ 6.00 GHz
GPU : NVIDIA GeForce RTX 4090
GPU Driver : nvidia (proprietary) 575.64

Program Version
Build: 12cdffcfc6c52291047010df3c398a7f291f410e / 0.0.0
2025-06-15 11:59:49 +0000 UTC

**Additional context**
no luqidctl or openrgb installed, in case of possible conflicts with OLH.

jurkovic-nikola, 2025-06-22:
It crashes on trying to Read data after a Write function. This points to low level communication issue with the slipstream device itself.

depcustodian, 2025-06-22:
That is strange, i've checked logs for slipstream and mouse, it seems they work fine. So, may be that is hid driver bug? Anyway, until i find any solution for this, is there any option to disable mouse initialization for OpenLinkHub for not to cause crashes?

depcustodian, 2025-06-22:
Yeah, cable is an option, buuut i don't like them, all my setup is done with minimal cable use. Still, even if i try to connect mouse directly to PC, i'll need at least 3m cable to put it right and clean through all my cable managment :D
But for testing purposes i'm already running mouse in cable mode. Will keep you in touch, how it works

depcustodian, 2025-06-22:
Oh, i haven't thought about that, will try!

depcustodian, 2025-06-22:
Well, that didnt last for long. It turned out, that not only mouse causes crash, but wired keyboard as well. The issue is the same
```
Jun 22 21:47:45 magnumopus OpenLinkHub[758]:         /home/kaldor/OpenLinkHub/src/devices/k70pro/k70pro.go:1205 +0x1233
Jun 22 21:47:45 magnumopus OpenLinkHub[758]: created by OpenLinkHub/src/devices/k70pro.(*Device).setDeviceColor in goroutine 264
Jun 22 21:47:45 magnumopus OpenLinkHub[758]:         /home/kaldor/OpenLinkHub/src/devices/k70pro/k70pro.go:1049 +0x675
Jun 22 21:47:45 magnumopus systemd[1]: OpenLinkHub.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Jun 22 21:47:45 magnumopus systemd[1]: OpenLinkHub.service: Failed with result 'exit-code'.
```
To be precise panic occured right after i started to customize rgb

depcustodian, 2025-06-22:
Flickering, but i suppose it might not be connected. It wasn't instat crash. ILH managed to work for some time

jurkovic-nikola, 2025-06-22:
Im running this on latest kernel on arch on 0.5.6 version with same device as you, and it works fine.

jurkovic-nikola, 2025-06-23:
Interesting, I would expect in stacktrace at least some concurrent mention. If it fixes it, great and make a PR.

jurkovic-nikola, 2025-06-29:
I've included your changes in 0.5.9

depcustodian, 2025-06-30:
> I've included your changes in 0.5.9

Thx, sorry for not keeping you up to date. I'm still in testing. there were few crashes, while i was using modified version. But i'm not sure, whether they were connected to the applied changes. Still, the current situation is this: i've made a clean install of OLH, modified `stats.go` as mentioned above. Result: clean run without crashes for several days already

jurkovic-nikola, 2025-06-30:
Thats good news. I'll also try to reproduce this on my side

closed 2025-08-30 by jurkovic-nikola