jurkovic-nikola/OpenLinkHub #143: K95 Platninum not responding at all even in bios.

Grumpy-Otter, 2025-06-26:
So I have a wierd issue, I have been trying this out on Bazzite and I tried openlinkhub it oppened and while toying with setting my keyboard stopped working completely, Im unable to do anything at all, even in bios, I noticed that if I plug the keyboard to different USB it started working but the moment I tried openlinkhub again, it bricked keyboard again.

If anyone knows what to do, I would love some help.

jurkovic-nikola, 2025-06-26:
Can you delete your keyboard json profile and restart service ? Profiles are located in `/opt/OpenLinkHub/database` in format of your device serial json file. You can get serial from WebUI.

Grumpy-Otter, 2025-06-26:
I will try later, I was trying out different light modes, and one I think said disabled, and thats when it stopped working

jurkovic-nikola, 2025-06-26:
How do you bind F13-24 in iCUE ? Only thing what I see there is that existing keys can be assigned to F13-24 also. 

[image]

Grumpy-Otter, 2025-06-26:
What I do is I bind f24 to my mouse button and then bind that as PTT in discord, f23 is my shadowplay that's bound to the macro keys on the keyboard etc...

Grumpy-Otter, 2025-06-26:
Ok, NVM found it, it was in different folder for me, I had to tinker a little to get it installed on Bazzite and profiles saved where I have extracted the openlinkhub archive, its all working now

closed 2025-06-26 by Grumpy-Otter