jurkovic-nikola/OpenLinkHub #145: [Feature Request] Tray Icons for battery life (remote devices)

shutdown4life, 2025-06-26:
Hey,

can you please create a tray icon for the battery life?
Would be also insane if it would show how much % the battery has. (iCUE had just low/middle/high/full

jurkovic-nikola, 2025-06-28:
Hey,

This can not be implemented in this project. OpenLinkHub runs as system application and under system user, which has no connection to user session and can not communicate with user space D-Bus interface. 

This will have to be separate application that will run in user space once user logs in.

closed 2025-06-28 by jurkovic-nikola

jurkovic-nikola, 2025-06-29:
[image]

This will be released as standalone app at https://github.com/jurkovic-nikola/openlinkhub-tray