jurkovic-nikola/OpenLinkHub #147: Support for other hwmon devices like Aquacomputer Quadro

lambertjosh, 2025-07-02:
**Describe the solution you'd like**
It would be great to support reading temperatures from other hwmon sensors than what is supported today. I have a Aquacomputer Quadro, which has a kernel driver. (See output below from sensors)

It would be great if this device could get picked up in the list of temperature sensors. I've tried to go through the configuration, but I haven't seen anything other than setting the cpu temp sensor which looks like it only supports a subset. 

My use case is that these are radiator fans (and some case fans) and I would like to use the liquid temp in a system that is not fully iCue.

**Additional context**

Sensor output from Quadro:

```
quadro-hid-3-f
Adapter: HID adapter
Fan 1 voltage:      12.06 V  
Fan 2 voltage:      12.06 V  
Fan 3 voltage:      12.04 V  
Fan 4 voltage:      12.06 V  
Fan 1 speed:       2353 RPM
Fan 2 speed:          0 RPM
Fan 3 speed:       1394 RPM
Fan 4 speed:          0 RPM
Flow speed [dL/h]:    0 RPM
Sensor 1:           +33.9°C  
Sensor 2:               N/A  
Sensor 3:               N/A  
Sensor 4:               N/A  
Virtual sensor 1:       N/A  
Virtual sensor 2:       N/A  
Virtual sensor 3:       N/A  
Virtual sensor 4:       N/A  
Virtual sensor 5:       N/A  
Virtual sensor 6:       N/A  
Virtual sensor 7:       N/A  
Virtual sensor 8:       N/A  
Virtual sensor 9:       N/A  
Virtual sensor 10:      N/A  
Virtual sensor 11:      N/A  
Virtual sensor 12:      N/A  
Virtual sensor 13:      N/A  
Virtual sensor 14:      N/A  
Virtual sensor 15:      N/A  
Virtual sensor 16:      N/A  
Fan 1 power:         0.00 W  
Fan 2 power:         0.00 W  
Fan 3 power:         3.44 W  
Fan 4 power:         0.00 W  
Fan 1 current:       0.00 A  
Fan 2 current:       0.00 A  
Fan 3 current:     286.00 mA 
Fan 4 current:       0.00 A  
pwm1:                   72%
pwm2:                  128%
pwm3:                   86%
pwm4:                  128%
```

jurkovic-nikola, 2025-07-02:
Yeah, that's possible. 0.6.0 will have it.

lambertjosh, 2025-07-02:
That's awesome, thanks @jurkovic-nikola. If you can point to a place where to add, I can try as well.

jurkovic-nikola, 2025-07-03:
It's already implemented. You'll have option to pick External Hwmon when creating new temperature profile and select wanted sensor. 

[image]