jurkovic-nikola/OpenLinkHub #152: Request for Corsair SABRE RGB PRO Gaming Mouse

Freaklimit, 2025-07-08:
Hello,

First of all, thank you for your work on this project—it's incredibly appreciated by users like me who rely on great device support.

I'd like to request support for the Corsair Sabre RGB PRO Gaming Mouse.

**Product link:**
[](https://www.corsair.com/us/en/p/gaming-mouse/ch-9303111-na/sabre-rgb-pro-champion-series-ultra-light-fps-moba-gaming-mouse-ch-9303111-na)

**Additional details:**
- The mouse is wired and currently connected through my monitor's USB hub, which is then connected to the PC via USB 3.0.
- Basic functionality like cursor movement and button clicks work as expected.
- I'm hoping to get support for extended features such as RGB lighting, DPI configuration, and polling rate adjustment, if possible.

Please let me know if you need any logs.

jurkovic-nikola, 2025-07-08:
Hello, please do a `lsusb -d 1b1c:` from terminal in order to get product ID

jurkovic-nikola, 2025-07-08:
@Freaklimit also link to Corsair website with mouse would be awesome.

Freaklimit, 2025-07-08:
> Hello, please do a `lsusb -d 1b1c:` from terminal in order to get product ID

Here is the output:
`Bus 003 Device 008: ID 1b1c:1b79 Corsair CORSAIR SABRE RGB PRO Gaming Mouse`

> [@Freaklimit](https://github.com/Freaklimit) also link to Corsair website with mouse would be awesome.

Sorry about the earlier message without the link! I used the link button in the markdown editor and didn’t realize it doesn’t work properly. Here’s the correct link to the Corsair website with the mouse:
https://www.corsair.com/us/en/p/gaming-mouse/ch-9303111-na/sabre-rgb-pro-champion-series-ultra-light-fps-moba-gaming-mouse-ch-9303111-na

jurkovic-nikola, 2025-07-08:
Thanks, will take some time to source this device.

closed 2025-08-02 by jurkovic-nikola