jurkovic-nikola/OpenLinkHub #155: Support for K65 PLUS Wireless

Absolemo, 2025-07-09:
**Describe the bug**
Hello everyone
I have switched from Windows to CachyOs. Now I would like to change the RGBs of my K65 PLUS Wireless. The Device Dashboard shows me everything (GPU,CPU,...) but unfortunately not my keyboard so I can't change my RGBs. Any advice?

Thanks

ChinaskiJr, 2025-08-02:
Hi @jurkovic-nikola ,
I have the same issue with k65 plus Wireless also. 

I'm on Ubuntu 25.04, running OpenLinkHub 0.6.0

The keyboard PID ( `2b11`) is not explicitely supported on the readme (only 2b10) : 
`Bus 003 Device 009: ID 1b1c:2b11 Corsair CORSAIR K65 PLUS WIRELESS Keyboard`

Here is the output of stdout.log, we can see that it does see our corsair keyboard. 
```
{"device":"CORSAIR K65 PLUS WIRELESS Keyboard","interface":0,"level":"info","message":"Processing device...","productId":11025,"serial":"AD00000017001000754DB64095E00000","timestamp":"2025-08-02T10:51:40+02:00"}
{"device":"CORSAIR K65 PLUS WIRELESS Keyboard","interface":0,"level":"info","message":"Processing device...","productId":11025,"serial":"AD00000017001000754DB64095E00000","timestamp":"2025-08-02T10:51:40+02:00"}
{"device":"CORSAIR K65 PLUS WIRELESS Keyboard","interface":0,"level":"info","message":"Processing device...","productId":11025,"serial":"AD00000017001000754DB64095E00000","timestamp":"2025-08-02T10:51:40+02:00"}
{"device":"CORSAIR K65 PLUS WIRELESS Keyboard","interface":0,"level":"info","message":"Processing device...","productId":11025,"serial":"AD00000017001000754DB64095E00000","timestamp":"2025-08-02T10:51:40+02:00"}
{"device":"CORSAIR K65 PLUS WIRELESS Keyboard","interface":0,"level":"info","message":"Processing device...","productId":11025,"serial":"AD00000017001000754DB64095E00000","timestamp":"2025-08-02T10:51:40+02:00"}
{"device":"CORSAIR K65 PLUS WIRELESS Keyboard","interface":1,"level":"info","message":"Processing device...","productId":11025,"serial":"AD00000017001000754DB64095E00000","timestamp":"2025-08-02T10:51:40+02:00"}
{"device":"CORSAIR K65 PLUS WIRELESS Keyboard","interface":2,"level":"info","message":"Processing device...","productId":11025,"serial":"AD00000017001000754DB64095E00000","timestamp":"2025-08-02T10:51:40+02:00"}
{"level":"info","message":"Starting USB monitor...","timestamp":"2025-08-02T10:51:40+02:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","message":"Scheduler file is found, checking for any upgrade...","timestamp":"2025-08-02T10:51:40+02:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","message":"Nothing to upgrade.","timestamp":"2025-08-02T10:51:40+02:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","message":"Dashboard file is found, checking for any upgrade...","timestamp":"2025-08-02T10:52:30+02:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","message":"Nothing to upgrade.","timestamp":"2025-08-02T10:52:30+02:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","message":"Dashboard file is found, checking for any upgrade...","timestamp":"2025-08-02T10:53:14+02:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","message":"Nothing to upgrade.","timestamp":"2025-08-02T10:53:14+02:00"}
```

Here is the config.json file 
```
{
    "debug": false,
    "listenPort": 27003,
    "listenAddress": "127.0.0.1",
    "cpuSensorChip": "",
    "manual": false,
    "frontend": true,
    "metrics": false,
    "memory": false,
    "memorySmBus": "i2c-0",
    "memoryType": 4,
    "exclude": [],
    "decodeMemorySku": true,
    "memorySku": "",
    "resumeDelay": 15000,
    "logFile": "",
    "logLevel": "info",
    "enhancementKits": "",
    "temperatureOffset": 0,
    "amdGpuIndex": 0,
    "amdsmiPath": "",
    "checkDevicePermission": true,
    "graphProfiles": false,
    "cpuTempFile": "",
    "ramTempViaHwmon": false
}
```

As you can see, the keyboard doesn't show : 
[image]

ChinaskiJr, 2025-08-02:
Here it is 

[image]

jurkovic-nikola, 2025-08-02:
Yeah, `2b11` is UK version of the device.

ChinaskiJr, 2025-08-02:
Strange, as you can see it has ISO french AZERTY layout.

jurkovic-nikola, 2025-08-02:
Whats the output of lsusb for dongle ?

ChinaskiJr, 2025-08-02:
TLDR; 

Try it with the USB dongle instead of the USB wire @Absolemo to see if it solves your problem

jurkovic-nikola, 2025-08-02:
I guess you're on French layout, right ?

ChinaskiJr, 2025-08-02:
> I guess you're on French layout, right ?

Absolutely