jurkovic-nikola/OpenLinkHub #156: For information: amd-smi: command not found

Hyourinmaru0, 2025-07-11:
For information for other users: If you execute `amd-smi static --asic --json` and terminal ouput says `amd-smi: command not found`, install the `amdsmi` package alongside the `rocm-core` package as dependency. After the install, the `amd-smi` package is located in `/opt/rocm/bin/` (or use `where amd-smi` to get the location) and now it can be executed to get the amdgpuindex.
In the README.md and the configuration help page on openlinkhub.dev, it's not clearly and described, the package is not installed from the very start and must be install manually.

I had to find that out because it didn't work for me and I first had to find out the name of the package and that it was part of ROCm and not installed.

jurkovic-nikola, 2025-07-11:
No it doesn't and it wont, since every package system can have different name. In addition, amd-smi package maintainer for for any distribution should take care of dependencies (like ROCm).

Hyourinmaru0, 2025-07-11:
Ahh, okay it makes sense. I didn't know, the package name is not standardize over package systems. I thought, the package name is the same for all package managers.

Nevermind... ^^"

Hyourinmaru0, 2025-07-11:
Apart from that: I appreciate your work and I'm exited for the future releases especially for the per-LED control feature. ^^