jurkovic-nikola/OpenLinkHub #157: Icue Profile Import Tool

pyrow14, 2025-07-11:
**Describe the solution you'd like**
Allow End user to import an exported .icueprofile. 
-Within the icue software on windows, a user can go and export the entire profile of leds, animations, fan curves, and device settings
-having the ability to import this file for this web GUI  to use would be amazing
-no expectations of being able to edit, but if the tool can read what was imported and use any profile imported  would scratch my itch

**Additional context**
Inside the iCue software we have the ability to export the .icueprofile which looks like json or xml for the icue tool to use in setting led patterns or such. Having an import feature could help users who want to port over the icue features that were configured previously on windows while also attempting to adopt linux

pyrow14, 2025-07-11:
Great work btw, since corsair is unlikely to ever support linux, im super thankful for smart folks like you to say F it, im doing what they wont!

jurkovic-nikola, 2025-07-11:
Hello, this was asked before and it's not possible to do it. This program and iCUE are just not compatible in such manner. They use XML for everything and this is using JSON for all configuration. I did examine XML from iCUE, and unfortunately there isn't tech documentation what each value in XML represents.

closed 2025-07-11 by jurkovic-nikola