jurkovic-nikola/OpenLinkHub #158: Duplicate RGB Profiles

KShivum, 2025-07-13:
**Describe the solution you'd like**
The ability to create a duplicate profile with different colors (I.E duplicate color shift so two different fans on a hub can have two different colors)

**Additional context**
Hi, I tried to add more color profiles to the rgb.json but the UI didn't seem to pick it up. But I want to be able to have each of my fans have different colors using the color shift profile but since you can only have one instance of the color shift profile and all of my fans are on one hub. It doesn't seem to be possible

jurkovic-nikola, 2025-07-13:
Hello, I'm aware of it. Duplication of profiles won't do much or editing rgb.json since this is just template from where actual profile is created for a device. RGB override is in works, and probably will be pushed out in 0.6.1 where you'll be able to override colors and speed for each device if needed.

jurkovic-nikola, 2025-07-17:
[image]

This will be released in 0.6.1. You will have option to override any RGB profile you assign to a device. Devices getting this:
- iCUE Link System Hub
- Commander Core + All AIOs using this hub
- Commander Core XT