jurkovic-nikola/OpenLinkHub #163: Add support for K70 Pro Mini Wireless

ryanfelder, 2025-07-27:
Wireless id:
Bus 003 Device 014: ID 1b1c:1ba6 Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver

Wired id:
Bus 003 Device 016: ID 1b1c:1bb6 Corsair CORSAIR K70 RGB PRO MINI SLIPSTREAM WIRELESS 60% Mechanical Gaming Keyboard

I would prefer to use this keyboard in wireless mode, but would be fine with programming it only while wired.
Please let me know any way I can assist.

ryanfelder, 2025-07-27:
Because I saw it in another 'add support' request, here's verbose output for wireless

`
lsusb -d 1b1c:1ba6 -v

Bus 003 Device 014: ID 1b1c:1ba6 Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver
Couldn't open device, some information will be missing
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1ba6 CORSAIR SLIPSTREAM WIRELESS USB Receiver
  bcdDevice            5.06
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR SLIPSTREAM WIRELESS USB Receiver
  iSerial                 3 81CD79BA184BEF92
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00a6
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength      85
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
[... 130 lines cut ...]
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength      90
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
`

jurkovic-nikola, 2025-07-27:
Thanks, will take some time to receive the device. Both modes will be available, same as any other Slipstream device here.

jurkovic-nikola, 2025-07-30:
Will be out in 0.6.1

[image]

ryanfelder, 2025-08-03:
I just installed this and I'm having a little trouble.

First, it didn't detect my device over wireless, but seems to work on wired.

Once connected up with wired, it has crashed twice while playing around with lighting effects, requiring a systemctl start to bring it back.

And some of the animations aren't behaving how I would expect.

**CircleShift, ColorPulse, ColorShift, ColorWarp, CpuTemperature, GpuTemperature, Off, Rainbow, Rotator, Static, Storm, Watercolor** -  all seem fine

**Circle** - First color is applied, second color is always black. Speed control works fine.
**Wave** - First color is applied, second color is always black. Speed control works fine.

**Flickering** - respects both colors, but in an unexpected way. meta keys are one color, the alphanumeric keys are another and they flicker between the assigned color and black. This might be intended.

**Spinner** - This one seems to light up each key individually, sequentially, following some pattern. Not an exciting mode. It feels like this one might not be working as intended.

Here are my crash logs 

```
Aug 03 15:10:54 ryan-desktop OpenLinkHub[1808604]: [Server] Running REST and WebUI on 127.0.0.1:27003. WebUI is accessible via: http://127.0.0.1:27003
Aug 03 15:22:46 ryan-desktop OpenLinkHub[1808604]: panic: runtime error: index out of range [201] with length 0
Aug 03 15:22:46 ryan-desktop OpenLinkHub[1808604]: goroutine 3089 [running]:
Aug 03 15:22:46 ryan-desktop OpenLinkHub[1808604]: OpenLinkHub/src/devices/k70pmWU.(*Device).setDeviceColor.func1(0xc004b23830?)
Aug 03 15:22:46 ryan-desktop OpenLinkHub[1808604]:         /app/0.6.1/src/devices/k70pmWU/k70pmWU.go:1601 +0x13a8
Aug 03 15:22:46 ryan-desktop OpenLinkHub[1808604]: created by OpenLinkHub/src/devices/k70pmWU.(*Device).setDeviceColor in goroutine 89
Aug 03 15:22:46 ryan-desktop OpenLinkHub[1808604]:         /app/0.6.1/src/devices/k70pmWU/k70pmWU.go:1462 +0x768
Aug 03 15:22:46 ryan-desktop systemd[1]: OpenLinkHub.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Aug 03 15:22:46 ryan-desktop systemd[1]: OpenLinkHub.service: Failed with result 'exit-code'.
Aug 03 15:22:46 ryan-desktop systemd[1]: OpenLinkHub.service: Consumed 21.919s CPU time, 408M memory peak.
```
and

```
Aug 03 15:54:54 ryan-desktop OpenLinkHub[2000309]: 2025/08/03 15:54:54 http: panic serving 127.0.0.1:51722: send on closed channel
Aug 03 15:54:54 ryan-desktop OpenLinkHub[2000309]: goroutine 114 [running]:
Aug 03 15:54:54 ryan-desktop OpenLinkHub[2000309]: net/http.(*conn).serve.func1()
Aug 03 15:54:54 ryan-desktop OpenLinkHub[2000309]:         /usr/local/go/src/net/http/server.go:1947 +0xbe
Aug 03 15:54:54 ryan-desktop OpenLinkHub[2000309]: panic({0xdb6ce0?, 0xf9a2d0?})
```

Also, after setting things up on wired, and then going back to wireless, OpenLinkHub no longer sees my keyboard, so it doesn't seem to be a one-time wired setup thing, it just seems like it doesn't support the wireless ID.

Finally, I was really hoping that OpenLinkHub would allow me to customize the colors on one of the built in effects.
There's an effect that lights up a key when you hit it, but the color is random. I was hoping to have the ability to set that to a static color. I don't see that mode at all in the lighting options in OpenLinkHub. If customizing that mode is possible, please let me know how. If it's not possible, no worries, and thank you for adding support!

jurkovic-nikola, 2025-08-03:
Hey, set debug to true in config.json file and try wireless mode. I'll need packet output.