jurkovic-nikola/OpenLinkHub #166: GPU Temperature is shown as 0.0 in multi-GPU system

AJWelchVontas, 2025-07-29:
**Describe the bug**
I installed OpenLinkHub a few days ago.
After installing a second GPU and putting in a liquid cooling loop, the web UI is showing both GPUs as a single temperature sensor and the temperature reading stays at 0.0. If I physically uninstall one of my GPUs, the web UI does display the GPU and the temperature properly. Please see attached screenshots.

**Expected behavior**
It should display two temperature sensors, one for each GPU, and the temperature should not read 0.0.

**Screenshots**

[image]
[image]

**Desktop (please complete the following information):**
 - OS: Debian 12

jurkovic-nikola, 2025-07-29:
Hello, thanks for reporting. I've never built it actually for multi GPU support, meh...

0.6.1 will bring following:

[image]

You'll have also ability to create temperature profiles from multi gpu selection:

[image]

AJWelchVontas, 2025-07-29:
Awesome, I'm looking forward to the next release! ^_^

closed 2025-08-02 by jurkovic-nikola

jurkovic-nikola, 2025-08-02:
Update to 0.6.1 and add following to config.json file

```json
  "nvidiaGpuIndex": [0,1],
```