jurkovic-nikola/OpenLinkHub #171: System76 has very low volume using the Void RGB Elite.

axiomtutor, 2025-08-02:
**Describe the bug**
The microphone volume is extremely low.

**Device / program info**
1. Content of `/opt/OpenLinkHub/config.json`
3. Output of `lsusb` command
4. Attach `/opt/OpenLinkHub/stdout.log` file.

1.
```
{
    "debug": false,
    "listenPort": 27003,
    "listenAddress": "127.0.0.1",
    "cpuSensorChip": "",
    "manual": false,
    "frontend": true,
    "metrics": false,
    "memory": false,
    "memorySmBus": "i2c-0",
    "memoryType": 4,
    "exclude": [],
    "decodeMemorySku": true,
    "memorySku": "",
    "resumeDelay": 15000,
    "logFile": "",
    "logLevel": "info",
    "enhancementKits": "",
    "temperatureOffset": 0,
    "amdGpuIndex": 0,
    "amdsmiPath": "",
    "checkDevicePermission": true,
    "graphProfiles": false,
    "cpuTempFile": "",
    "ramTempViaHwmon": false
}
```

2.
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 5986:9102 Acer, Inc BisonCam,NB Pro
Bus 001 Device 006: ID 1b1c:0a55 Corsair CORSAIR VOID ELITE Wireless Gaming Dongle
Bus 001 Device 004: ID 8087:0033 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

3.
[stdout.log](https://github.com/user-attachments/files/21559323/stdout.log)

**Expected behavior**
Open sound monitor, and have a larger audio signal when speaking clearly into the mic.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Pop!_OS
 - Device Firmware: Result of `uname -r` is `6.12.10-76061203-generic`
 - Program Version [e.g. 0.3.9]: 0.6.1 I think (couldn't find the info in the documentation, but I just installed it yesterday).  

**Additional context**
I installed this and used it on a different computer running Linux Mint, and it works perfectly.  

When I open the UI on my System76 computer (the one with the low volume), it looks like this:

[image]

jurkovic-nikola, 2025-08-02:
Hm, how can I help you with this ?

VOID ELITE is not even on supported device list. You need to check somewhere else why mic volume is low.

axiomtutor, 2025-08-02:
Ahh, sorry, didn't check the list!  It worked on Mint, so assumed it was supported.  

Never mind, thanks for developing the project.