jurkovic-nikola/OpenLinkHub #186: Sync All Devices RGB

jbmerrill1954, 2025-08-11:
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Hello,
I can't seem to find a way to sync all devices at once with a color scheme / choice.  Perhaps I am overlooking it but would like to request that this feature be added so that I can just select one color effect and have it apply to all devices at once rather than setting them all individually.  Is this possible?

Otherwise, I am really enjoying this application and thank you for the great work you have done on it.
B.

**Additional context**
Add any other context or screenshots about the feature request here.

jbmerrill1954, 2025-08-11:
Also, my Corsair K95 RGB platinum keyboard is not being detected.   I will try to find what info you require for debugging this, but if you could point me in the right direction, I would appreciate it.
B.

jbmerrill1954, 2025-08-11:
[image]

jbmerrill1954, 2025-08-12:
@bryan  lsusb -d 1b1c:
Bus 001 Device 003: ID 1b1c:1b5c Corsair CORSAIR NIGHTSWORD RGB Gaming Mouse
Bus 001 Device 029: ID 1b1c:1b89 Corsair CORSAIR K95 RGB PLATINUM XT Mechanical Gaming Keyboard
Bus 001 Device 030: ID 1b1c:1b9b Corsair CORSAIR MM700 RGB Gaming Mousepad
Bus 001 Device 031: ID 1b1c:0c3f Corsair iCUE LINK System Hub
Bus 001 Device 036: ID 1b1c:0c4e Corsair CORSAIR iCUE LINK AIO LCD Screen Module
Bus 001 Device 037: ID 1b1c:0c3f Corsair iCUE LINK System Hub

 Tue 12 Aug - 02:10  ~ 
 @bryan  

[stdout.log](https://github.com/user-attachments/files/21732687/stdout.log)

jurkovic-nikola, 2025-08-20:
Get on Discord https://discord.gg/mPHcasZRPy it will be much faster there.

jbmerrill1954, 2025-08-20:
I got on Discord, but have not yet claimed my account because I don't know what I am going to say there.  Do I just refer to issue #186?   Thanks and sorry if I am not doing this right.

jurkovic-nikola, 2025-08-20:
I've already pinged you on private, so message will popup once you set your account.

jurkovic-nikola, 2025-08-30:
K95 Platinum XT will be added in next release, since i'm waiting on delivery of the device.

jbmerrill1954, 2025-09-15:
> K95 Platinum XT will be out in 0.6.5, or you can compile latest code from main.

Great, I will try compiling and let you know.  Thank you very much.Best regards,BryanSent from my iPhone

closed 2025-09-17 by jurkovic-nikola