jurkovic-nikola/OpenLinkHub #205: Changes to Watercolor RGB color profile

akimbler, 2025-09-02:
I had previously been running v0.6.0 and upgraded to v0.6.2 last night and noticed that the Watercolor lighting profile appears to have been changed between versions without documentation. The new profile appears stuttery and seems to be missing a color or something because it doesn't look close to the default Corsair profile unlike the one from v0.6.0. Is this an intended change?

This is the v0.6.0 version

https://github.com/user-attachments/assets/2ddc14e5-5edd-4a1f-9d5c-1d2ca06276d0

This is the version on v0.6.1+

https://github.com/user-attachments/assets/a900ec81-0b87-4c38-b20c-883341705498

jurkovic-nikola, 2025-09-02:
This is because all previous versions have a bug in speed slider, it was in reverse. All of this was changed thru 0.6.1 and 0.6.2. In addition of slider bug, there was timer bug where if you ran different RGB effects, they would reset in different time, it wasn't synchronized at all.

jurkovic-nikola, 2025-09-02:
Indeed, you're right. There was hue, saturation and position change. Let me dig this thru...

akimbler, 2025-09-02:
Awesome! Just built from Main with the new commit and it's working perfectly. Thanks for getting on this so quickly!

jurkovic-nikola, 2025-09-03:
This will be out for packages in 0.6.3

closed 2025-09-03 by jurkovic-nikola