jurkovic-nikola/OpenLinkHub #210: No Devices Detected?

honsontran, 2025-09-07:
Hello all,

I'm running on CachyOS (Arch-based), and I've been installing this via the AUR and also through manual means via this repo. I am running into an issue where I cannot get my Corsair mouse to detect on the dashboard. What logs do I need to share to help debug the issue?

[image]

honsontran, 2025-09-07:
So, it appears I needed to plug in the device directly via USB, as a wireless connection was not able to change the settings.

honsontran, 2025-09-07:
Edit, I currently have the slipstream wireless dongle plugged in, as well as the mouse itself:
```
$ lsusb -d 1b1c:

Bus 001 Device 007: ID 1b1c:2b00 Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver
Bus 001 Device 010: ID 1b1c:2b22 Corsair CORSAIR SCIMITAR ELITE WIRELESS SE Gaming Mouse
```

It seems as if `2b00` is not detected, but my settings for `2b22` work fine.

jurkovic-nikola, 2025-09-07:
Slipstream currently supports only multipoint connections. You can fix this just by re-pairing your mouse with the dongle via iCUE. This will be resolved in couple of next versions.

closed 2025-09-14 by jurkovic-nikola