jurkovic-nikola/OpenLinkHub #220: Corsair VENGEANCE PRO RGB PRO DDR5 4x48GB 4800 MT/s

N3L3K, 2025-09-18:
Hello,

I have several peripherals from the brand Corsair and everything is recognized perfectly in the dashboard. However, my DDR5 RAM is not, even after trying some of the recommended manipulations.

What information can I provide you with so you can help me?

Thanks in advance

N3L3K, 2025-09-18:
Kernel : 6.16.7-2-cachyos

Result with  i2cdetect -l (**which one is my device ???** )

i2c-0	i2c       	NVIDIA i2c adapter 1 at 1:00.0  	I2C adapter
i2c-1	i2c       	NVIDIA i2c adapter 2 at 1:00.0  	I2C adapter
i2c-2	i2c       	NVIDIA i2c adapter 3 at 1:00.0  	I2C adapter
i2c-3	i2c       	NVIDIA i2c adapter 4 at 1:00.0  	I2C adapter
i2c-4	i2c       	NVIDIA i2c adapter 5 at 1:00.0  	I2C adapter
i2c-5	i2c       	NVIDIA i2c adapter 6 at 1:00.0  	I2C adapter
i2c-6	smbus     	SMBus I801 adapter at 0000:00:1f.4	SMBus adapter
i2c-7	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-8	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-9	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-10	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-11	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-12	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter

jurkovic-nikola, 2025-09-19:
Solved over Discord. SPD Write Lock was enabled in BIOS, and i2c permissions were missing.

closed 2025-09-19 by jurkovic-nikola